brocaar/chirpstack-network-server #113: Nonofficial RU 864-869 ISM band

Vini74, 2016-12-26:
Is it possible to add experimental ISM band for Russian Federation?

Ch# 	Freq	                Modulation	        duty cycle	default channels
1	864,10 MHz	MuliSF 125kHz	0,1% or LBT	yes
2	864,30 MHz	MuliSF 125kHz	0,1% or LBT	yes
3	864,50 MHz	MuliSF 125kHz	0,1% or LBT	yes
4	864,70 MHz	MuliSF 125kHz	0,1% or LBT	
RX2	864,90 MHz	SF12 125kHz	        0,1% or LBT	
5	868,85 MHz	MuliSF 125kHz	-	
6	869,05 MHz	MuliSF 125kHz	-

brocaar, 2016-12-27:
Could you share any documents / drafts describing this experimental ISM band?

Vini74, 2016-12-29:
Draft specifications for LoRaWAN Regional Parameters attached. 

-----

brocaar, 2016-12-29:
Hi @Vini74 it seems that you can't attach files to a github issue by mail. Could you send this file to my e-mail address (see http://github.com/brocaar)? Or attach it to this issue using the GitHub web-interface?

brocaar, 2017-01-02:
Hi @Vini74 because of NY my previous message have been slipped through. Could you please send me an e-mail with the pdf attached or add it using the GitHub web-interface? It seems that attachements are stripped when replying to the issue by mail. Thanks and an happy new year!

Vini74, 2017-01-02:
[LoRaWAN_RU-868-Regional_Parameters_v0_1.pdf](https://github.com/brocaar/loraserver/files/681343/LoRaWAN_RU-868-Regional_Parameters_v0_1.pdf)

brocaar, 2017-01-03:
Cheers! I'll re-open the issue so that I can keep track of all feature requests until they are implemented :-)

Vini74, 2017-01-17:
[LoRaWAN_RU-868-Regional_Parameters_v1.0.pdf](https://github.com/brocaar/loraserver/files/710668/LoRaWAN_RU-868-Regional_Parameters_v1.0.pdf)

Dear Orne,
 
We have completed additional tests in the allowed frequency bands and concluded that in the 864.0 MHz range chip 1301 has significant decrease in receiver sensitivity. That is why we have to reassign frequencies as follows:
864.1 MHz -> 864.5 MHz
864.3 MHz -> 864.7 MHz
864.5 MHz -> 864.9 MHz
RX2 - 869.05 MHz
 
Please find updated documentation in attached file. Could you please make necessary modifications at your earliest convenience? Thank you.
Also please take my apologies for the fuss.

brocaar, 2017-01-29:
@Vini74 I've updated this in the master branch. Could you test these changes?

Vini74, 2017-02-01:
Your changes look like good! I shall try build this revision.

Vini74, 2017-02-01:
https://github.com/brocaar

brocaar, 2017-02-28:
@Vini74 can I close this issue (the latest draft is in the latest LoRa Server release)?

commit 4a289274 (2017-01-08): Update band configuration + docs.

Closes #113, #50 and #49.
closed 2017-02-28 by brocaar