brocaar/chirpstack-network-server #114: timestamp in MQTT

Pbars, 2016-12-28:
Hello Brocaar
how do I publish the timestamp in MQTT. the following message is what I get on screen:
{"devEUI":"514d3857b73bb24e","rxInfo":[{"mac":"1dee0ad91e56bbb5","rssi":-54,"loRaSNR":9.5}],"txInfo":{"frequency":868100000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":7},"adr":true,"codeRate":"4/5"},"fCnt":4,"fPort":8,"data":"Aahde5YAhRys"}
thanks for the support

brocaar, 2017-01-08:
Hi @Pbars, I'm not sure if I get your question / understand what you're trying to achieve?

aurorasmarenghi, 2017-01-18:
Hi @brocaar,
I am a collegue of @Pbars. We have a lorank 8 gateway (without gps module). Our trouble is the disappeared of timestamp in the paylod. Do you know how to fix it?
Many thanks

paolo

brocaar, 2017-01-18:
@aurorasmarenghi ah now I understand. The time field is only filled when there is an accurate time source available on your gateway (read GPS).

aurorasmarenghi, 2017-01-19:
@brocaar , many thanks, you have dissipated my doubts

closed 2017-01-27 by brocaar