brocaar/chirpstack-network-server #116: DevNonce has already been used

t-stricker, 2017-01-05:
Hi brocaar,

any Idea about this log message from loraserver?
`level=error msg="processing rx packet error: application server join-request error: rpc error: code = 3 desc = DevNonce has already been used" data_base64="AMwHJO7LJRpSAWk1ZTI4MTFMdxb9Ibs="`

t-stricker, 2017-01-05:
I have a guess. There was a second join request incoming from the mote. Then the server recognises double session. The first session wasn't send to the gateway. I changed some configuration I did for the "antenna-gain" now its working again. Not sure if it was like this. I will let you know If I know more :)

brocaar, 2017-01-07:
As re-using a `DevNonce` value would compromise the security of your node, each time your node sends a join-request (OTAA), the application server adds the `DevNonce` of this join-request to the list of that node. In this case it happened that your node re-used the same value or when you have multiple gateways, it could be that one gateway has a high network latency and because of this it was seen as a re-transmission.

t-stricker, 2017-01-12:
Hi Broocar, thanks for your answer.
it seams to be any latency problem. It's occurring not regularly.
You can close this issue

richard61230, 2017-02-24:
Hi  Brocaar  
I  have another question about DevNonce  if node happen run out of  DevNonce for join request(OTAA) ) whether LoRa server will not allow node  join successfully until clear DevNonce of LoRa server recorded?
How can i clear  node's DevNonce  of LoRa server recorded 
Thank you

I used LoRa server version : -0.12.5
       lora appserver version: 0.1.4

brocaar, 2017-02-24:
Actually that is a good point. I think the only way to flush the set of `DevNonce` values in a safe way would be by changing the `AppKey`. The `DevNonce` is there so that somebody can't replay a join-request, however, a new `AppKey` would also avoid replay attacks. I'll think about this. Maybe it is as simple as when updating the `AppKey`, reset `DevNonce` set.

richard61230, 2017-02-26:
Hi Brocaar, thanks for your answer 
It is glad to hear your idea , i have not figured out to  flush DevNonce set safely
It's a good way to implement so that it means now All of release loRa server versions , there have not a  way to flush DevNonoce . Will it  be  implemented in future versions ?

brocaar, 2017-02-27:
Yes, I think I'll implement this in a future version :-)

For now, clearing the `used_dev_nonces` column (`node` table in the database) would do the trick.

richard61230, 2017-03-01:
Hi Brocaar , thanks for your help again  :-)
I'll try it andi look forward to your good news

carlberg74, 2017-05-02:
Hi Brocaar, 
I have similar issue.  (one gateway only)
When I clear the used_dev_nonces I can see the device try to join again.
but after a couple of seconds, I start getting "join-request DevNonce has already been used" again.
I don't have anyway to change the AppKey on the device.
I guess this is because the device is sending a join and not waiting long enough (nor completing the first join) before its sends the next join? 

Any tip what I can do to get around this?

brocaar, 2017-05-05:
When you're unable to change the `AppKey` and fix the firmware (so that it doesn't send out so many join-requests), I think there is not much you can do, except to clear out the used dev-nonces. However note that this introduces a security risk!

brocaar, 2017-05-16:
In the next release of LoRa App Server, the used dev-nonces will be reset on AppKey change.

rnicolas, 2018-01-13:
Hi brocaar,
I've come here for the same reason, a doubt in the DevNonce and a theorical limit of 65,536 possible joins per device without changing the AppKey. I understand the idea if there's a change on the AppKey then the DevNonce is reset to 0 and for that reason a node could send other 65,536 joins more..., but to be able to do that, the AppKey must be generated and sent/updated also to the device, but that's not a common behavior written on LoRaWAN specification (correct me if I'm wrong), and that should be done in an automatic way.
Instead of that, it's specified that some number of DevNonces are stored, but it's not specified a clear number of how many of them should be stored. With a single PRNG, the rule of thumb is that a 50% chance of a collision happens after generating the square root of the range. So, if you were generating 16-bit random numbers, your collision probability reaches 50% after about 300 numbers generated because the square root of 2^16 is 256.
So, why don't just keep track of a limited number of devNonces and not all devNonces? I know that's a possible security flaw, but that's how is told in the specification.

brocaar, 2018-01-13:
@rnicolas please note that this has been updated in LoRaWAN 1.1 (specification). Instead of a random nonce it is specified to be an incremental nonce.

For < 1.1 do you have an idea which nonces should be stored when not storing all nonces? The advantage I see by storing all the nonces is that a collision would cause a join failure, which should trigger the device to re-try (using a different nonce). The result is that a nonce would never ever been re-used. The more nonces yes would mean more collisions and more collisions would meen that a device would need more join-request attempts to hit an unused nonce.

rnicolas, 2018-01-15:
@brocaar 
Hi again, thank you for the information, I was looking at specification 1.0.1 instead of 1.1; and yes, in the 1.1 I think the problem is solved using the rejoin procedure. But in the specification 1.0.1 there is no rejoin procedure and what is said is this: _DevNonce  is  a  random  value. For  each  end-­device,  the  network  server  keeps  track  of  a certain  number  of  DevNonce  values  used  by  the  end-­device  in  the  past,  and  ignores  join requests  with  any  of  these  DevNonce  values  from  that  end-­device._ 
So..., there's no number defined. Now I was trying to read the quantity of nonces that I already have registered in the database, but they are stored in bytea format, and when I read using `select encode(used_dev_nonces, 'base64') from node;` I get a bunch of letters that I don't know how to handle. So I'm not able know (because of a lack of aknowlege of reading this data) how many nonces I have stored, with that value I could say when I experienced a decrease in performance on the nodes trying to do joins. But I know that if all nonces are stored (using 1.0.1 specifications), in a while the nodes won't be able to join the network anymore because all the nonces in some moment will have been used.

zhangjqxc, 2018-05-25:
@brocaar Hello Brocaar, I am quite interested to know why LoRaWAN 1.1 specification decides to use an incremental DevNonce instead of a random nonce specified in the old specification. Any special consideration?

brocaar, 2018-05-28:
I think because it results in less collisions / the device doesn't have to keep track of every used DevNonce. Instead by knowing the last used DevNonce, it directly knows that all previous DevNonces already have been used. I don't think having a random DevNonce is an advantage, it just have to be unique which also works by incrementing it.

brocaar, 2018-08-13:
@dzwer unfortunately that is how it is specified for LoRaWAN 1.0.x. Keeping track of only the last X still allows for replay-attacks. However, you could always implement a cleanup SQL query in case needed.

closed 2018-08-13 by brocaar

Sam-Eshun, 2018-09-10:
@brocaar I wish to find out in case there's a reset how does the lora server know ther's a reset in order for the counter to start from 0 at the server side. According to the specification, a reset by ABP devices will make the counter start from 0(both the device and the server). How does the server know ther's a reset at the end-device.

Sam-Eshun, 2018-09-10:
I wish to find out in case there's a reset how does the lora server know ther's a reset in order for the counter to start from 0 at the server side. According to the specification, a reset by ABP devices will make the counter start from 0(both the device and the server). How does the server know ther's a reset at the end-device.

brocaar, 2018-09-10:
You should **not** start from 0 after a power-cycle:

[image]

Sam-Eshun, 2018-09-10:
@brocaar Thank you for the response..I undertsand, in ABP devices, when the device is reset the counter at the device side start from 0. My question, how does the server know the device has been reset so the counter at the server side must also start from 0 as described in the specification.

brocaar, 2018-09-10:
> in ABP devices, when the device is reset the counter at the device side start from 0

This should not happen. The device **must** keep track of used nonces (specification: *...a DevNonce value SHALL NEVER be reused for a given JoinEUI value...*).

Sam-Eshun, 2018-09-10:
@brocaar Ok. My question is in reference to LoRaWAN 1.0 ABP devices.

brocaar, 2018-09-10:
In the case of LoRaWAN 1.0 your DevNonce should not start at 0 and increment (LoRaWAN 1.0.3 specs):

[image]

Note that each time when your device resets, the chance that you'll hit a DevNonce which has already been used will increase. That is why this meganism has changed in the 1.1 specs.

> My question is in reference to LoRaWAN 1.0 ABP devices.

Please note that in ABP node, you're not dealing with DevNonces! You need to keep track of the frame-counter values. The DevNonce is for join-requests / -accepts. Also here your device needs to keep track of the frame-counter values and in case of a power-cycle make sure that this is not lost.

If you want to continue this discussion, please use the forum: https://forum.loraserver.io/. This information can be helpful to others and GitHub issues are not the best way to build a knowledge base.

MarekNovakACRIOS, 2021-10-05:
Hi all,

We found out this to be a practical limitation of LoRaWAN 1.0.x  - our device do regularily re-joins to avoid problems with up/down counters + send from time to time confirmed uplink to detect, that there is some issue with security context and that they should execute a rejoin... This shows in practice as a good solution and we have deployed quite a lot of devices and have no issues so far. BUT - we are slowly approaching the limit of 65535 with this approach and even if this will come in several (a few) years, we need to find a fix... 

After thinking about it, we found a solution, that we drop recorded nonces, which are older than 100 days:
```
delete from public.device_activation where (DATE_PART('day', now()) - DATE_PART('day', created_at)) > 100
```
Practically that means, that a replay attacker, who record the packets for 100 days and then starts sending the recorded packets 100 days later would be able to force the network server to drop the security context for given device and the device would have to rejoin and then he would be able to attack again etc.. but he would newer get control over the device.

It is a security problem, but it is maybe better than having no way to join. Also updating AppKey looks like a much better approach, but so far, we don't have this implemented in deployed devices (but we might deploy it, since we have OTAA there and the devices are Class C, mains powered). 

Did any of you implement the approach of updating AppKey after some time?