brocaar/chirpstack-network-server #117: Invalid frame count error - relax frame counter enabled

t-stricker, 2017-01-12:
Hello, I'm receiving this error in the loraserver logs:
`processing rx packet error: invalid FCnt or too many dropped frames`

I'm using ABP for my mote (Semtech LoRaMote) and configured the relax frame counter.
The mote is emitting each ~15 seconds uplink packets. Sometimes its working, sometimes I get the error.

jamiechard, 2017-01-16:
Hello. The following may be unrelated, but I saw similar issue recently. 
My Node sent data to the server at an unknown Data Rate (i.e. a data rate that didn’t exist in the DataRates table). This resulted in the downlink.SendDataDownResponse() called from handleCollectedDataUpPackets() in internal/uplink/data.go failing. However, since the ns.FCntUp had been incremented a few lines above, the NS FCntUp ended up being one in ahead of the Node FCntUp, which produced the “invalid FCnt or too many dropped frames” error message. I’m guessing that the ns.FCntUp may need to be decremented again if the uplink ACK or downlink ACK fail? However, I worked around the issue by adding the “missing” Data Rate to the DataRates table.

brocaar, 2017-01-18:
@t-stricker do you have more than one gateway? It could be that one of your gateway has a high latency. The server waits a short while to de-duplicate the same frame received by multiple gateways. When it continues and then receives the same frame from a gateway with a high latency, it will return in the above error.

@jamiechard was the data-rate missing because you are testing with different data-rates than specified by the LoRaWAN specs or should they have been there? The so-far known configs are defined here: https://github.com/brocaar/lorawan/tree/master/band

t-stricker, 2017-01-19:
@brocaar no it is only one gateway. So shouldn't be a latency issue.
I'm routing the gateway traffic a quite long distance, over multiple routers. But it is still inside my LAN. So the response from the network server should be fast. 

Maybe configuring the "Receive window delay" in the app-server frontend can help out?

brocaar, 2017-01-21:
The only thing I could think of is that the frames are not processed in the order they were transmitted. E.g. when the `FCnt` is validated, it is fine to go from 10 to 12 (11 will be considered as lost), however when 11 comes in afterwards it raises the error that the `FCnt` is invalid. What you could do is keep an eye on the `fCnt` field in https://docs.loraserver.io/lora-app-server/mqtt-topics/#receiving. Then when the above error occurs, debug if there were any gaps.

systemmonkey42, 2017-02-01:
I am having the same problem, and I think the code is suspect

> func ValidateAndGetFullFCntUp(n NodeSession, fCntUp uint32) (uint32, bool) {
>     // we need to compare the difference of the 16 LSB
>     gap := uint32(uint16(fCntUp) - uint16(n.FCntUp%65536))
>     if gap < common.Band.MaxFCntGap {
>         return n.FCntUp + gap, true
>     }
> 

I find the fCntUp (752) is less than "n.FCntUp" (1537), and the gap becomes negative, however the forced conversion to uint32 results in a gap of 64751 which then results in a "not ok" verdict.

I think you might need to calculate the absolute value before converting to unsigned or comparing with the threshold value.

Does this make sense? I hope I'm not misunderstanding the purpose of the Validate function.

Thanks

brocaar, 2017-02-02:
Hi @systemmonkey42 thanks for your feedback. Just to summarise:

Received `FCnt: 752`, expected `FCnt: 1537`.

To me it seems that this must be invalidated, as it is a possible replay-attack. The gap calculated is not the gap backwards, but forwards (in case of missed frames / packet loss) and taking a `FCnt` rollover (16 bit space) into account.

To validate this calculated gap, you can make the following calculation:

`(1537+64751)%65536 = 752`

See here the list of testcases for this function:
https://github.com/brocaar/loraserver/blob/master/internal/session/session_test.go#L125

closed 2017-04-25 by brocaar