brocaar/chirpstack-network-server #118: Not Work downlink

sabas1080, 2017-01-21:
Hi @brocaar 

I have two nodes one  with RN2903 and other with RFM95
The two work perfect in OTAA, doing the join and the TX  but we cant receive the downlink from our server

We have tried with TTN server and everything works good, we did the join and TX and then we tried the Downlink and it worked, is it any difference between ttn server and this version of LoRa server to consider? 

Thanks

brocaar, 2017-01-21:
Hi @sabas1080 could you add logs of LoRa Gateway Bridge, LoRa Server and LoRa App Server to this issue? Could you also let me know which gateway + packet_forwarder version you're using? That makes it easier for me to help you debugging this issue :-)

alwint3r, 2017-01-23:
Hi @brocaar, I also have the same issue here. I use  iC880A Concentrator and https://github.com/TheThingsNetwork/packet_forwarder as the packet forwarder.

Here's few logs from app server & gateway bridge:

App server:
```
time="2017-01-22T11:58:32Z" level=info msg="downlink queue item deleted" id=22 
time="2017-01-22T11:58:32Z" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0004a30b001b2d10 
time="2017-01-22T11:59:09Z" level=info msg="handler/mqtt: publishing data-up payload" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/rx" 
time="2017-01-22T11:59:09Z" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0004a30b001b2d10 
time="2017-01-22T11:59:25Z" level=info msg="handler/mqtt: data-down payload received" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/tx" 
time="2017-01-22T11:59:25Z" level=info msg="downlink queue item enqueued" dev_eui=0004a30b001b2d10 id=23 
time="2017-01-22T11:59:46Z" level=info msg="handler/mqtt: publishing data-up payload" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/rx" 
time="2017-01-22T11:59:47Z" level=warning msg="queue item discarded as it exceeds max payload size" dev_eui=0004a30b001b2d10 max_payload_size=51 payload_size=60 reference=1485086365600 
time="2017-01-22T11:59:47Z" level=info msg="downlink queue item deleted" id=23 
time="2017-01-22T11:59:47Z" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0004a30b001b2d10 
time="2017-01-22T12:00:10Z" level=error msg="handler/mqtt: mqtt connection error: EOF" 
time="2017-01-22T12:00:25Z" level=info msg="handler/mqtt: connected to mqtt broker" 
time="2017-01-22T12:00:25Z" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx" 
time="2017-01-22T12:01:01Z" level=info msg="handler/mqtt: publishing data-up payload" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/rx" 
time="2017-01-22T12:01:01Z" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0004a30b001b2d10 
time="2017-01-22T12:01:38Z" level=info msg="handler/mqtt: publishing data-up payload" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/rx" 
time="2017-01-22T12:01:38Z" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0004a30b001b2d10 
time="2017-01-22T12:02:15Z" level=info msg="handler/mqtt: publishing data-up payload" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/rx" 
time="2017-01-22T12:02:15Z" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0004a30b001b2d10 
time="2017-01-22T12:02:52Z" level=info msg="handler/mqtt: publishing data-up payload" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/rx" 
time="2017-01-22T12:02:52Z" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0004a30b001b2d10 
time="2017-01-22T12:03:29Z" level=info msg="handler/mqtt: publishing data-up payload" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/rx" 
time="2017-01-22T12:03:29Z" level=info msg="data-down item requested by network-server, but queue is empty" dev_eui=0004a30b001b2d10 
time="2017-01-22T12:23:36Z" level=info msg="handler/mqtt: data-down payload received" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/tx" 
time="2017-01-22T12:23:36Z" level=info msg="downlink queue item enqueued" dev_eui=0004a30b001b2d10 id=24 
time="2017-01-22T12:23:49Z" level=info msg="handler/mqtt: data-down payload received" topic="application/c99df5f485d7ab21/node/0004a30b001b2d10/tx" 
time="2017-01-22T12:23:49Z" level=info msg="downlink queue item enqueued" dev_eui=0004a30b001b2d10 id=25 
```

```
time="2017-01-22T11:58:02Z" level=info msg="gateway: received udp packet from gateway" addr=115.178.193.41:53420 protocol_version=1 type=PullData 
time="2017-01-22T11:58:02Z" level=info msg="gateway: sending udp packet to gateway" addr=115.178.193.41:53420 protocol_version=1 type=PullACK 
time="2017-01-22T11:58:32Z" level=info msg="gateway: received udp packet from gateway" addr=115.178.193.41:38444 protocol_version=1 type=PushData 
time="2017-01-22T11:58:32Z" level=info msg="gateway: rxpk packet received" addr=115.178.193.41:38444 data="QGoaASaAAgAB0/PcKz4iD8T/zkRgonxb" mac=fb10b827ebe4f14e 
time="2017-01-22T11:58:32Z" level=info msg="gateway: rxpk packet received" addr=115.178.193.41:38444 data="QGoaASaAAgAB0/PcKz4iD8T/zkRgonxb" mac=fb10b827ebe4f14e 
time="2017-01-22T11:58:32Z" level=info msg="backend: publishing packet" topic="gateway/fb10b827ebe4f14e/rx" 
time="2017-01-22T11:58:32Z" level=info msg="backend: publishing packet" topic="gateway/fb10b827ebe4f14e/rx" 
time="2017-01-22T11:58:32Z" level=info msg="gateway: sending udp packet to gateway" addr=115.178.193.41:38444 protocol_version=1 type=PushACK 
time="2017-01-22T11:59:03Z" level=info msg="gateway: received udp packet from gateway" addr=115.178.193.41:40507 protocol_version=1 type=PullData 
time="2017-01-22T11:59:03Z" level=info msg="gateway: sending udp packet to gateway" addr=115.178.193.41:40507 protocol_version=1 type=PullACK 
time="2017-01-22T11:59:09Z" level=info msg="gateway: received udp packet from gateway" addr=115.178.193.41:37396 protocol_version=1 type=PushData 
time="2017-01-22T11:59:09Z" level=info msg="gateway: rxpk packet received" addr=115.178.193.41:37396 data="QGoaASaAAwABwim/V7sz31JS/nk+gBLP" mac=fb10b827ebe4f14e 
[... cut ...]
time="2017-01-22T12:03:29Z" level=info msg="gateway: sending udp packet to gateway" addr=115.178.193.41:60906 protocol_version=1 type=PushACK 
time="2017-01-22T12:04:03Z" level=info msg="gateway: received udp packet from gateway" addr=115.178.193.41:42042 protocol_version=1 type=PullData 
time="2017-01-22T12:04:03Z" level=info msg="gateway: sending udp packet to gateway" addr=115.178.193.41:42042 protocol_version=1 type=PullACK 
```

Unfortunately I accidentally deleted the lora server logs because I re-created the container. 

Thanks in advance!

brocaar, 2017-01-23:
The LoRa App Server warning might explain your issue 😉 

```
time="2017-01-22T11:58:32Z" level=warning msg="queue item discarded as it exceeds max payload size" dev_eui=0004a30b001b2d10 max_payload_size=51 payload_size=60 reference=1485086303624
```

LoRaWAN has a max-payload size per data-rate. Please see the "LoRaWAN regional parameters" spec for these values per data-rate. Also note that in the latest LoRa Server version, you can specify if your setup uses a repeater or not. In the latter case, the max-payload can be increased (this info is also in the LoRaWAN regional parameters" spec.

(I'm not sure if TTN has a validation on this, this might explain the difference in behaviour)

alwint3r, 2017-01-24:
Thanks for your response! So the problem should be in the application and not on the lora server and gateway, right?

I'm also wondering whether the `reference` should be unique or not.

brocaar, 2017-01-24:
@alwint3r yes I suppose so. You could use the same reference for every payload, however it might be confusing at your side as it is used in case of errors and acknowledgements (in case of confirmed data down)

alwint3r, 2017-01-24:
@brocaar Thanks for the info. I'll tell you later whether it's solved or not.

alwint3r, 2017-01-24:
I think the downlink issue is solved for me. I used shorter reference and reduced the payload size and it's working fine now. Thank you!

sabas1080, 2017-01-25:
Hi 

Update to

lora server v0.13.3
lora gateway bridge v2.1.2
lora app server v0.2.0

and work in RFM95

Thanks @brocaar

closed 2017-01-25 by sabas1080

dpmendes, 2018-02-07:
Hi, @alwint3r !

How did you manage to find these logs?
I'm running loraserver on a Raspberry Pi 3 with an image generated by loraserver-pi-gen.
Even though everything is working fine, I can't find any logs.
All directories in /var/log/<program> are empty or have empty log files.

Thanks!