brocaar/chirpstack-network-server #119: loraserver on RPI3 "transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused"

gesaleh, 2017-01-22:
Hi 
I managed to install it and run on my raspberry 3 but when the loraserver is stated i'm getting this error
which part need to be changed on the firewall (iptable)
8001 8080 443 1700 ?

```
 ./loraserver --net-id 147474 --band EU_863_870
INFO[0000] starting LoRa Server                          
band="EU_863_870" docs="https://docs.loraserver.io/" 
net_id=147474 
version=0.13.3-3-g81f8ff8
INFO[0000] setup redis connection pool                   url="redis://localhost:6379"
INFO[0000] backend/gateway: connecting to mqtt broker    server="tcp://localhost:1883"
INFO[0000] connecting to application-server              ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
INFO[0000] no network-controller configured
INFO[0000] starting api server                           bind="0.0.0.0:8000" ca-cert= tls-cert= tls-key=
INFO[0000] backend/gateway: connected to mqtt server
INFO[0000] backend/gateway: subscribing to rx topic      topic="gateway/+/rx"
INFO[0000] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}
INFO[0001] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}

```

gesaleh, 2017-01-22:
I also instaaled lora-app server and bridge
i had to install postgres and create a DB called loraserver 
still getting the same issue and when i start the server app i get this

```
./lora-app-server --postgres-dsn  postgres://postgres:postgres@localhost/loraserver?sslmode=disable &
 INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.2.0-4-g6a33bcf
INFO[0000] connecting to postgresql
INFO[0000] setup redis connection pool
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
{"pid":18800,"hostname":"raspberrypi","level":30,"time":1485121983272,"msg":"connected","client":{"id":"c2d2ce9d-314f-40cc-a7dd-e0832b32ecc5"},"v":1}
new client c2d2ce9d-314f-40cc-a7dd-e0832b32ecc5
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] handler/mqtt: connected to mqtt broker
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
INFO[0000] starting application-server api               bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key=
WARN[0000] client api authentication and authorization is disabled (set jwt-secret to enable)
INFO[0000] starting client api server                    bind="0.0.0.0:8080" tls-cert= tls-key=
FATA[0000] open : no such file or directory
```

gesaleh, 2017-01-23:
adding information about version of the app server in fact it doesn't start VERSION:
   0.2.0-4-g6a33bcf
i keep having this error can't open

pablocaviglia, 2017-01-23:
hi @gesaleh, the problem is that you haven't added the certificate and private key for the application server.

The easiest way is to generate a pem file.
Then you have two ways to make the server read those files

1) Setting environment variables with the paths

      HTTP_TLS_CERT=/app/cert/lora-app-server.pem
      HTTP_TLS_KEY=/app/cert/lora-app-server.pem

2) Passing the paths through init parameters to lora-app-server program

   --http-tls-cert value  /app/cert/lora-app-server.pem
   --http-tls-key value  /app/cert/lora-app-server.pem

If that works then your log will not show any more last line (FATA[0000] open : no such file or directory)

brocaar, 2017-01-27:
This is related to https://github.com/brocaar/lora-app-server/issues/15

closed 2017-01-27 by brocaar

gesaleh, 2017-01-29:
Thank you I'll generate the pem files and continue the testing 
I'm trying to have it working with my GW (now running against TTN)

MohammedBENHADINE, 2017-02-15:
so gesaleh,it's running now?!