brocaar/chirpstack-network-server #120: Confirmed downlink data error

mittsopach, 2017-01-26:
Hello, I have problem with Confirmed downlink data from LoRa server to LMIC node. Here is MACPayload sequence from join to error:

JOIN REQUEST FROM NODE LMIC:
0,166,134,121,244,0,0,0,0
JOIN ACCEPTED FROM LORA SERVER:
1,85,156,4,0,0,0,0,
SEND CONFIRMED UPLINK FROM NODE LMIC:
128,85,156,4,0,0,1,0,1,29,3,148,69,8,49,142,234,235,81,32,17,238,249,166
RECIEVED ACK IN DOWNLINK FROM LORA SERVER:
96,85,156,4,0,32,3,0,0,0
SET CONFIRMED DATA TO QUEUE DOWNLINK (MQTT):
{ "reference": "data1234", "confirmed": true, "devEUI": "0000000010111216", "fPort": 99, "data": "QAAB" }
SEND CONFIRMED UPLINK FROM NODE LMIC:
128,85,156,4,0,0,3,0,1,29,3,148,69,8,49,142,234,235,81,32,17,238,249,166
RECIEVED ACK IN DOWNLINK FROM LORA SERVER WITH CONFIRMED DOWNLINK DATA:
160,85,156,4,0,32,4,0,99,64,0,1
SEND ACK FOR DOWNLINK DATA FROM NODE LMIC:
64,85,156,4,0,32,5,0,0,0
ERRORS FROM LORA SERVER (MQTT):
{"devEUI":"0000000010111216","type":"DATA_UP_FCNT","error":"invalid FCnt or too many dropped frames (server_fcnt: 6, packet_fcnt: 5)"}
{"devEUI":"0000000010111216","type":"DATA_UP_FCNT","error":"invalid FCnt or too many dropped frames (server_fcnt: 6, packet_fcnt: 5)"}
{"devEUI":"0000000010111216","type":"DATA_UP_FCNT","error":"invalid FCnt or too many dropped frames (server_fcnt: 6, packet_fcnt: 5)"}

and after theese errors all data from LORA SERVER are incorrect and LMIC library return: Invalid downlink, window=RX1. I tried it with checked/unchecked option Enable relax frame-counter, with the same result.

Can you check it, or any idea please ?

LoRaWan Server 0.13.3
LoRaWan App Server 0.2.0

brocaar, 2017-01-27:
I'm not sure if this is a bug at the LoRa Server side. What happens is that your node sends a frame with `FCnt: 5`, however the most recent frame that was processed had `FCnt: 6`, so it sees the frame as a replay-attack.

mittsopach, 2017-01-27:
I don't think so. 

Now I tested it again, and on LoRa server was FcntUp=4 and FcntDn=4 and then you can see:

SET CONFIRMED DATA TO QUEUE DOWNLINK (MQTT):
{ "reference": "data1234", "confirmed": true, "devEUI": "0000000010111216", "fPort": 99, "data": "QAAB" }
SEND CONFIRMED UPLINK FROM NODE LMIC:
128,170,218,146,1,0,4,0
RECIEVED ACK IN DOWNLINK FROM LORA SERVER WITH CONFIRMED DOWNLINK DATA:
160,170,218,146,1,32,4,0,99,64,0,1
SEND ACK FOR DOWNLINK DATA FROM NODE LMIC (this is automatically sent by LMIC library):
64,170,218,146,1,32,4,0
and again 2 ERRORS FROM LORA SERVER (MQTT):
{"devEUI":"0000000010111216","type":"DATA_UP_FCNT","error":"invalid FCnt or too many dropped frames (server_fcnt: 5, packet_fcnt: 4)"}
{"devEUI":"0000000010111216","type":"DATA_UP_FCNT","error":"invalid FCnt or too many dropped frames (server_fcnt: 5, packet_fcnt: 4)"}
I think that LMIC library shoul working properly with MAC protocol

Any idea pls ? And how then working Enable relax frame-counter feature ?

brocaar, 2017-01-27:
Please see the error message, that will give you a clue about which FCnt was received, and which FCnt was expected:

```
invalid FCnt or too many dropped frames (server_fcnt: 5, packet_fcnt: 4)
```

The relax frame-counter is only for ABP nodes. When the node resets (e.g. power cycle), its FCnt is reset to `0`, while the server expects for example `123`. When relax frame-counter is enabled, it will reset the FCnt to `0` at the server-side, instead of complaining about the invalid FCnt.

One observation form the above log: could it be that your node is re-sending the same frame? Please try uplink / downlink without `confirmed: true`, see if that works. If not, could you please post your LoRa Gateway Bridge, LoRa Server and LoRa App Server logs?

mittsopach, 2017-01-27:
Downlink unconfirmed is OK. But I need to confirm it. I think that problem could be, that ack of downlink data from LMIC not increasing fcnt:

```
SEND CONFIRMED UPLINK FROM NODE LMIC:
128,170,218,146,1,0,4,0
RECIEVED ACK IN DOWNLINK FROM LORA SERVER WITH CONFIRMED DOWNLINK DATA:
160,170,218,146,1,32,4,0,99,64,0,1
SEND ACK FOR DOWNLINK DATA FROM NODE LMIC (this is automatically sent by LMIC library):
64,170,218,146,1,32,4,0
```

But I'm using standard LMIC library v 1.5. Everything else working properly ...

mittsopach, 2017-01-27:
Solved. Solution is programatically increase FcntUp after confirmed downlink is recieved. Thank you...

closed 2017-01-27 by mittsopach