brocaar/chirpstack-network-server #125: Error in Compiling

Umair-Anwar, 2017-02-07:
I have followed the instructions to compile with docker compose.
but i always get an error 'start_linux_init.go (178) exec format error'

I'm using raspbian on raspberry pi 3 (armv7)

brocaar, 2017-02-07:
I have no experience with Docker on ARM, but it looks like the container that you're using is either not for ARM or you're not compiling for the ARM architecture.

ivajloip, 2017-02-18:
Speaking of Docker + ARM + Golang - [this][1] comes to mind :)

[1]: https://hub.docker.com/r/acklio/golang-armhf/

closed 2017-04-25 by brocaar

ivajloip, 2017-04-25:
I had missed that issue. If you still need some inspiration, you can check out [this][1] Dockerfile that generates mosquitto for ARMv7 :) The result can be found [here][2].

[1]: https://bitbucket.org/acklio/docker-mosquitto/src/9cbbb2319c3b1a3cbeb788910b6e6b870d636552/armhf/Dockerfile?at=master&fileviewer=file-view-default
[2]: https://hub.docker.com/r/acklio/mosquitto/tags/