brocaar/chirpstack-network-server #126: transport: http2Server.HandleStreams received bogus greeting from client

khush-bhatia, 2017-02-07:
I see the following message repeated several time in loraserver logs. The lora-app-server is up and running and the http interface to lora-app-server is also working. 

loraserver[27297]: time="2017-02-07T15:21:57-05:00" level=info msg="transport: http2Server.HandleStreams received bogus greeting from client: \"\\x16\\x03\\x01\\x00\\xcd\\x01\\x00\\x00\\xc9\\x03\\x03\\x91ϕd|\\\\\\xfe\\xf1uG\\x16f\\xcc\""

brocaar, 2017-02-08:
Are you sure you're the only one connecting to your LoRa Server instance? Could it be that somebody else is scanning the open ports and guessing protocols for example?

khush-bhatia, 2017-02-08:
That explains it. Thanks

closed 2017-02-08 by khush-bhatia

aeb-1, 2020-09-26:
@khush-bhatia 
can u explain how u solved this

pls!!