brocaar/chirpstack-network-server #127: bug report~~code error

yzk0281, 2017-02-08:
Hi brocaar, there must be a code error in fhdr.go:
func (c *FCtrl) UnmarshalBinary(data []byte) error {
	if len(data) != 1 {
		return errors.New("lorawan: 1 byte of data is expected")
	}
	c.fOptsLen = data[0] & ((1 << 3) ^ (1 << 2) ^ (1 << 1) ^ (1 << 0))
	c.FPending = data[0]&(1<<4) > 0
	c.ACK = data[0]&(1<<5) > 0
	c.ADRACKReq = data[0]&(1<<6) > 0
	c.ADR = data[0]&(1<<7) > 0
	return nil
}

it should be:
func (c *FCtrl) UnmarshalBinary(data []byte) error {
	if len(data) != 1 {
		return errors.New("lorawan: 1 byte of data is expected")
	}
	c.fOptsLen = data[0] & ((1 << 3) ^ (1 << 2) ^ (1 << 1) ^ (1 << 0))
	c.FPending = (data[0]&(1<<4)) > 0
	c.ACK = (data[0]&(1<<5)) > 0
	c.ADRACKReq = (data[0]&(1<<6)) > 0
	c.ADR = (data[0]&(1<<7)) > 0
	return nil
}

yzk0281, 2017-02-08:
In C/C++, this seems to be a bug
but in go, it is not，  why?

brocaar, 2017-02-08:
Well, Go isn't C/C++ ;-) Anway if you think the behaviour is incorrect, could you open an issue at https://github.com/brocaar/lorawan (that is the repo for the `lorawan` package)?

There are tests too to validate the behaviour of this code:
https://github.com/brocaar/lorawan/blob/master/fhdr_test.go#L72

brocaar, 2017-02-18:
@yzk0281 do you need more input on this / do you still think this is a bug? Can this issue be closed else?

closed 2017-03-15 by brocaar