brocaar/chirpstack-network-server #128: How to create Users in LoRa App server

kamran000, 2017-02-08:
I have installed LoRa server and LoRa App server , accessed App server through http , but there is only option of adding devices , how can i create users in it?

brocaar, 2017-02-08:
This is currently not yet implemented, but is definitely something I'll be working on soon. Could you please create a feature request at https://github.com/brocaar/lora-app-server so you can track the progress (as this feature should be part of LoRa App Server, not LoRa Server)?

closed 2017-02-08 by brocaar

kamran000, 2017-02-08:
yeah sure. Thanks it will be very helpful

kamran000, 2017-02-08:
How to create feature request?

brocaar, 2017-02-08:
I meant, could you create an issue describing the feature request for the above repo :-)

kamran000, 2017-02-08:
ok :-)