brocaar/chirpstack-network-server #129: COPY the source code into the Docker

drasko, 2017-02-09:
Here https://github.com/brocaar/loraserver/blob/master/Dockerfile we are missing this part: https://github.com/mainflux/lora-gateway-bridge/blob/master/Dockerfile#L14-L15

brocaar, 2017-02-11:
Actually, it should be removed there 😉 A Docker volume is used to share the code between the host and the container (so that you're able to develop on your host while having the latest version of the code within your container. See:

https://github.com/brocaar/loraserver/blob/master/docker-compose.yml#L5

drasko, 2017-02-11:
@brocaar this is good approach for development, but now at all good approach for deployment. I do not want to git-clone the code in order to do the deployment. And I surely do not to build every time upon deployment (`make serve` target depends on `make build`).

What I want is just to automatically pull the images from DockerHub and deploy the composition with a simple `docker-compose start`.

Maybe you can have separate `docker-compose-develop`?

brocaar, 2017-02-11:
Note that these `Dockerfile` files were never intended for deployments, as they are quite big. For deployment you would ideally use a `FROM scratch` container with just the binary (which would be a few MB).

drasko, 2017-02-11:
Actually, this is what [Alpine Linux](https://alpinelinux.org/) is for, and in general I would prefer that you used `FROM golang:1.7-alpine` - that should make an image just a few MBs...

brocaar, 2017-02-16:
Could you elaborate on the advantage of Alpine Linux vs `FROM scratch`? Note that the `golang:1.7-alpine` is already 241MB:

```
$ docker images

REPOSITORY              TAG                 IMAGE ID            CREATED             SIZE
golang                  1.7-alpine          f7da70df716c        2 weeks ago         241 MB
```

drasko, 2017-02-16:
@brocaar maybe you are right and `FROM scratch` could be a good solution in Golang case: http://blog.xebia.com/create-the-smallest-possible-docker-container/

I had in my mind that `scratch` was just an empty baremetal container without rootfs and that you had to add Busybox. But from what I can see here: https://www.iron.io/microcontainers-tiny-portable-containers/, Apline Linux is good when your app needs some OS dependencies. However, since Go binaries can be statically compiled it looks like `scratch` would be sufficient, and probably better solution.

I am only affraid about `dockerize` script which I use in order for the app to wait other services, but I guess that in production code this can be solved by changing the app itself to timeout and retry...

drasko, 2017-02-16:
A word of warning for the reference - I think "Security" chapter here is worth examining: https://alpinelinux.org/about/

_Alpine Linux was designed with security in mind. The kernel is patched with an unofficial port of grsecurity/PaX, and all userland binaries are compiled as Position Independent Executables (PIE) with stack smashing protection. These proactive security features prevent exploitation of entire classes of zero-day and other vulnerabilities._

brocaar, 2017-02-17:
But when using the Go binary inside an Alpine linux container, it won't use any of that?... You would still be running the kernel of the host system which basically executes the binary from the container in a separate name-space. As Go would come self-contained / no external dependencies, it would not use any of the userland binaries provided by Alpine. 

That is why in think that `FROM scratch` is a better idea, as it is basically an empty container / layer. Since Docker 1.5 it would not even create a new layer: https://hub.docker.com/_/scratch/.

drasko, 2017-02-17:
@brocaar yes, as I wrote above I think you are right and that `FROM scratch` is better candidate here.

brocaar, 2017-02-17:
Ah sorry, was just looking at your last message ;-) I'll have a look at the `dockerize` script later to understand what it does / what needs to be changed in the code in order to support this.

ymkins, 2017-03-13:
My 5 cent: 
I use Ubuntu based Docker container with release binaries because of Alpine use musl instead of gnu libc.
It's not hard to build custom release, but I prefer monitor the project issues and exclude the builder in case of deployment troubles.
So, we can clarify the building process in docs.

drasko, 2017-03-27:
@ymkins if we go with full-blown Docker, I would rather suggest CentOS. But I sgree with @brocaar that `FROM scratch` should be used - once it is well controled and understood there should be no problems. And if problems noted, they should be solved Docker community rather then workarounding the problems.

KMax, 2017-03-27:
Created images for loraserver and lora-app-server, also docker-compose.yml with default settings for my needs. It based on Ubuntu 16.04 and the pre-compiled binaries. 

Repo: https://github.com/ceratop/loraserver-docker

P.S. @drasko Thanks for pointing to `dockerize` script.

drasko, 2017-03-27:
Based on corrections from @KMax, CentOS flavor can be found here: https://github.com/mainflux/loraserver-docker

brocaar, 2017-05-04:
Not long ago Docker announced multi-stage builds. I haven't had the time to look into it, but as I understand it, it allows you to have multiple stages in the Dockerfile. Thus a "fat" container compiling a binary, which gets inserted into a "tiny" container. I have to look into it, but it could be interesting!

brocaar, 2017-08-24:
@drasko I'm currently in the process of setting up automated Docker builds. In the end I choose Alpine Linux instead of Scratch because it is easier to include CA certificates inside the container. Thanks again for all the input! All Docker containers will appear at: https://hub.docker.com/u/loraserver/.

I've changed the `Dockerfile` into https://github.com/brocaar/loraserver/blob/master/Dockerfile.

brocaar, 2017-08-25:
I'll close this issue. For feedback on the provided Docker images, see https://forum.loraserver.io/t/docker-containers/73.

closed 2017-08-25 by brocaar