brocaar/chirpstack-network-server #130: Receive message from gateway with lower quality.

siscia, 2017-02-10:
Maybe related to #78 

We are working with a small number of gateway, however it seems that the message we receive from the `application/` MQTT endpoint is the packet received from the gateway with the lowest rssi and the lowest SNR.

And when I say lowest I mean really lower.

From gateway A I receive rssi on -112 and SNR of 1.2 and from gateway B I receive rssi of -89 and SNR of 7.8.

What I would expect is too see the message coming from the gateway B, however it comes from the gateway A, actually the payload are exactly the same, it is normal?

brocaar, 2017-02-10:
You should see all gateways which received the same frame in the MQTT endpoint. Are you sure there is only one item in `rxInfo`?

Note that LoRa Server waits a certain time to "de-duplicate" frames received by multiple gateways. It will only process these frames once, but the MQTT endpoint should give you the metadata of all gateways. It could be a latency issue too, see the `DEDUPLICATION_DELAY` setting https://docs.loraserver.io/loraserver/configuration/.

siscia, 2017-02-10:
Yes, I can see only one element in the `rxInfo` field.

I attached a couple of screen shot so you can see yourself, do you think it is me doing something wrong?

[image]
[image]

brocaar, 2017-02-11:
Hmm, could you maybe also share some LoRa Server logs? There should indeed be two `rxInfo` elements. I'll do some testing here too just to make sure that no regression was introduced during the last changes.

brocaar, 2017-02-18:
I've done some testing with two gateways, this is my result (which seems correct?):

```
gateway/1234567887654321/rx

{
	"rxInfo": {
		"mac": "1234567887654321",
		"time": "0001-01-01T00:00:00Z",
		"timestamp": 3967222532,
		"frequency": 868300000,
		"channel": 1,
		"rfChain": 0,
		"crcStatus": 1,
		"codeRate": "4/5",
		"rssi": -53,
		"loRaSNR": 8.8,
		"size": 15,
		"dataRate": {
			"modulation": "LORA",
			"spreadFactor": 10,
			"bandwidth": 125
		}
	},
	"phyPayload": "gFC1AwYAAgABV1qjf2Xf"
}

gateway/1dee08d0b691d149/rx

{
	"rxInfo": {
		"mac": "1dee08d0b691d149",
		"time": "0001-01-01T00:00:00Z",
		"timestamp": 1136348684,
		"frequency": 868300000,
		"channel": 1,
		"rfChain": 1,
		"crcStatus": 1,
		"codeRate": "4/5",
		"rssi": -41,
		"loRaSNR": 7.8,
		"size": 15,
		"dataRate": {
			"modulation": "LORA",
			"spreadFactor": 10,
			"bandwidth": 125
		}
	},
	"phyPayload": "gFC1AwYAAgABV1qjf2Xf"
}

application/3/node/0101010101010101/rx 

{
	"applicationID": "3",
	"applicationName": "serial-test",
	"nodeName": "uartsbee",
	"devEUI": "0101010101010101",
	"rxInfo": [{
		"mac": "1dee08d0b691d149",
		"rssi": -41,
		"loRaSNR": 7.8
	}, {
		"mac": "1234567887654321",
		"rssi": -53,
		"loRaSNR": 8.8
	}],
	"txInfo": {
		"frequency": 868300000,
		"dataRate": {
			"modulation": "LORA",
			"bandwidth": 125,
			"spreadFactor": 10
		},
		"adr": false,
		"codeRate": "4/5"
	},
	"fCnt": 2,
	"fPort": 1,
	"data": "qrs="
}
```

VanXungUIT, 2017-06-02:
Hello I want see data. Just like that. Can you help me? . I have done configure node information in gateway.

siscia, 2017-06-03:
@VanXungUIT, please describe what you didn't understand in the documentation so that it is possible to improve it.

closed 2017-06-03 by siscia