brocaar/chirpstack-network-server #131: Install loraserver on windows

ShahabR, 2017-02-13:
Hi,
Is there any help for installing lora server on windows os? In linux it talks about configuring lora server after installation, how about on windows? Thanks

hecko, 2017-02-13:
One way would be to use Docker.

brocaar, 2017-02-13:
Note that you can pass all the configuration as arguments. I don't have any experiences with setting up windows services, but if you run `loraserver.exe --help` on the command-line, you'll see all available arguments :-)

ShahabR, 2017-02-14:
Can you explain more about using docker in this case @hecko ( I know a little about docker) ? Thanks.
I've developed several windows service in .net language but I'm not familiar with this type of deployment( I use to install service with a batch file and... ) , anyway I will try using --help argument as you said, thanks brocaar.

closed 2017-07-25 by brocaar