brocaar/chirpstack-network-server #135: LoraServer cannot recognize mtype!

phatnt199, 2017-02-28:
I have forward data from lorawan gateway to my application, but mtype is unconfirmed, please help me with this problem!

Lora-server log: INFO[11085] backend/gateway: rx packet received          
INFO[11085] packet(s) collected                           dev_eui=4786e6ed00290028 gw_count=1 gw_macs=b827ebfffe9e04f1 mtype=UnconfirmedDataUp
INFO[11085] rx info sent to network-controller            dev_eui=4786e6ed00290028
INFO[11085] node-session saved                            dev_addr=cf426001

Lora-app-server log: INFO[11061] handler/mqtt: publishing data-up payload      topic="application/1/node/4786e6ed00290028/rx"
INFO[11061] data-down item requested by network-server, but queue is empty  dev_eui=4786e6ed00290028

MQTT log: application/1/node/4786e6ed00290028/rx {"applicationID":"1","applicationName":"SampleApp","nodeName":"TH_Sensor","devEUI":"4786e6ed00290028","rxInfo":[{"mac":"b827ebfffe9e04f1","rssi":-71,"loRaSNR":-4.5}],"txInfo":{"frequency":433800000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":true,"codeRate":"4/5"},"fCnt":7451,"fPort":8,"data":"AbRtbQUA//+7"}

brocaar, 2017-02-28:
What is the actual issue?

phatnt199, 2017-02-28:
How can I decode the data and save to the database?

brocaar, 2017-02-28:
Your (base64 encoded) data is: `AbRtbQUA//+7`. You need to base64 decode that in order to get the bytes that your sent :-)

closed 2017-03-01 by phatnt199