brocaar/chirpstack-network-server #139: ADR

kuripiiritus, 2017-03-09:
Hei

Probably not the right place to ask, but are you planning on adding ADR support (creating a network controller).

Best regard,
Kuldar

brocaar, 2017-03-09:
An experimental version of ADR has already been implemented for the EU 868 ISM band. I'm planning to add support for the other bands too, but I need to do some research on that first :-)

kuripiiritus, 2017-03-09:
Hei

Now that you mentioned it we installed the latest version, and there it is. 

`The interval (of frames) after which the network-server will ask the node to change data-rate and / or TX power if it can change to a better data-rate or lower TX power. Setting this to 0 will disable ADR.
`
Will be performing some tests on it.

BR,
Kuldar

closed 2017-03-12 by brocaar