brocaar/chirpstack-network-server #141: Node-session does not exist or invalide fcnt or mic

dylancollaud, 2017-03-13:
Hi Brocaar, 
I've install node-red and and your server (lora-gateway-server, loraserver, lora-app-server)
I've a gateway with ic880a. In the lora-gateway-bridge i see the lora message. 

But in the journal of loraserver i've this error : 
Mar 13 22:36:26 ttn-gateway-Collaud loraserver[1534]: time="2017-03-13T22:36:26+01:00" level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic" data_base64="QNUhASaADgAB/LoqYwGHlEyJuufU4ht4Zqo="

And i don't know what is the problem. 

Could you help me ?

sabas1080, 2017-03-14:
Hi

Is your node in ABP or OTAA? If ABP has already created the data?

In ABP remember to restart your fcnt

dylancollaud, 2017-03-14:
Hi,
My note is OTAA. I've tested with an other node and I've the same issue.
The node arrive to connect but when i send a uplink, I've this error.

brocaar, 2017-03-14:
@dylancollaud given the error message, I think you're using the 0.15.0 version. Before this error message was a bit different (showing the frame-counters), this is not possible anymore as a single `DevAddr` might be used by multiple nodes. So in order to match the correct node(-session), we need to validate the `MIC`. See #56 for some background.

Two questions:
* did the OTAA join work?
* could you compare the `FCnt` of your node in both LoRa App Server as your node itself?

dylancollaud, 2017-03-14:
Yes i've use le 0.15.0 version.
Yes the OTAA works, my device are connected and on the wep-app I've the Network session key.

For the last question, I dont understand where I can found the FCnt ?

brocaar, 2017-03-14:
In LoRa App Server when you edit a node, there is a "Show activation" button. It will show you the details known by the server.

From your node it depends on the node and firmware you're using. Did this error occur after you've reset the node?

dylancollaud, 2017-03-14:
Lora App Server 
Device address (generate)
06cb7154
Network session key
c1ee6f2daf68622c787ba3cfe8105120
Application session key
d6172443d4d6716c19345736473a1784

Node
DeviceAddress
5C0C00A0
Network session key
9EC1C0FA675E74E1F083E30030C5B01
Application session key
4E606B7B1A60B2488FE1B098DD76B1D

All are different .. I thing this isn't correct ?

brocaar, 2017-03-14:
That might explain why the node-session could not be found. Could it be caused by the node reset? E.g. because of the reset, the node reverted it's settings to the settings stored in the persistent memory. An OTAA activation might note be stored in an persistent way depending the firmware. 

Btw, a more secure way is to perform a new OTAA after a node reset. That way you don't need the relax frame-counter mode and you're safe against replay attacks.

dylancollaud, 2017-03-14:
I've installed the lora-gateway-bridge, loraserver and lora-app-server and all worked last week. 
After that, i've some problem with the ic880a and I reinstall all applications. 

And now, it doesn't work and it's stange.. 
I've the same issue on two devices (Dragino and Dongle UsSB Lora ACW-usb atim ARM-N8-LW)
When i reset my node, the device adresse doesn't change, it's stange. You probably rigth my node restore default device adresse. But when the node start, it will make an OTAA and save the new device adress and other key given by the gateway ?

dylancollaud, 2017-03-14:
I found ! ... 
I made a mistake, i configure in my gateway two lora-gateway-bridge. This is not possible in OTAA I'think.
thank you for your help ! You made awsome job !

closed 2017-03-14 by dylancollaud

RCOdev, 2017-03-28:
Hi everyone, I get the same error (processing rx packet error), "Node-session does not exist or invalide fcnt or mic".
I'm using ABP mode, the node is activated (seen in lora-app-server log) but loraserver give me this error.
Do you have an idea where this is come from?
Thanks!

Deepdone, 2017-03-30:
@RCOdev @brocaar i also get the same error log when a node(OTAA) worked a moment! I cannot but reset the node!

blunderbuss1, 2018-09-11:
@dylancollaud can you provide the code where you use the ATIM ACW USB Key ? Thanks in advance.

dylancollaud, 2018-09-12:
@blunderbuss1  I wrote a mini app on Windows in C# to control the device via Serial port with standarts commands. But this app is not finished and if you want try it, contact me in private. dylan.collaud(at)gmail.com