brocaar/chirpstack-network-server #147: Two node with same device eui will cause payload data abnormal

ckt1010, 2017-03-27:
Hi,
I found a strange problem today. 
When I use node device eui which is "0x4786e6ed00380057", payload parse ok. But if I use another node with same device eui, the first node still can receive data, but payload parse abnormal. Please refer log below:

>  gateway/aa555a0000000101/rx {"rxInfo":{"mac":"aa555a0000000101","time":"0001-01-01T00:00:00Z","timestamp":3705995188,"frequency":471500000,"channel":4,"rfChain":0,"crcStatus":1,"codeRate":"4/5","rssi":-4,"loRaSNR":10.5,"size":25,"dataRate":{"modulation":"LORA","spreadFactor":12,"bandwidth":125}},"phyPayload":"gNURwgaAAAACHscuvo2mQjcY6eDmRsMaRQ=="}
Mar 27 19:53:01 iZbp13xpaqt4mvtq1tt73nZ node[27584]: application/70b3d57ed000140d/node/**4786e6ed00380057**/rx {"devEUI":"4786e6ed00380057","rxInfo":[{"mac":"aa555a0000000101","rssi":-4,"loRaSNR":10.5}],"txInfo":{"frequency":471500000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":true,"codeRate":"4/5"},"fCnt":0,"fPort":2,"data":"AAAAAAAA//8AAAAA"}
Mar 27 19:53:02 iZbp13xpaqt4mvtq1tt73nZ node[27584]: gateway/aa555a0000000101/tx {"txInfo":{"mac":"aa555a0000000101","immediately":false,"timestamp":3706995188,"frequency":471500000,"power":14,"dataRate":{"modulation":"LORA","spreadFactor":12,"bandwidth":125},"codeRate":"4/5","iPol":null},"phyPayload":"YNURwgYgAADklLFd"}

> Mar 27 19:53:10 iZbp13xpaqt4mvtq1tt73nZ node[27584]: gateway/aa555a0000000101/rx {"rxInfo":{"mac":"aa555a0000000101","time":"0001-01-01T00:00:00Z","timestamp":3714762692,"frequency":471700000,"channel":5,"rfChain":0,"crcStatus":1,"codeRate":"4/5","rssi":-49,"loRaSNR":2.8,"size":25,"dataRate":{"modulation":"LORA","spreadFactor":12,"bandwidth":125}},"phyPayload":"gE0GbAeAAwACDqV+7FQX9I3yzt4aKKJvvQ=="}
Mar 27 19:53:10 iZbp13xpaqt4mvtq1tt73nZ node[27584]: application/70b3d57ed000140d/node/**4786e6ed00380057**/rx {"devEUI":"4786e6ed00380057","rxInfo":[{"mac":"aa555a0000000101","rssi":-49,"loRaSNR":2.8}],"txInfo":{"frequency":471700000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":true,"codeRate":"4/5"},"fCnt":3,"fPort":2,"data":"SyvbZEbu1bFrVKQ5"}
Mar 27 19:53:10 iZbp13xpaqt4mvtq1tt73nZ node[27584]: gateway/aa555a0000000101/tx {"txInfo":{"mac":"aa555a0000000101","immediately":false,"timestamp":3715762692,"frequency":471700000,"power":14,"dataRate":{"modulation":"LORA","spreadFactor":12,"bandwidth":125},"codeRate":"4/5","iPol":null},"phyPayload":"YE0GbAcgAwBPaVAI"}

You can see two packet from different node which use same device eui. Second packet  joined at first, and first packet joined at next.
The Most strange things is that verification about MIC value has no problem. And two node will receive different ACK which devaddr is same as they got before. 
Did I miss something?
Thanks for your reply!

brocaar, 2017-03-27:
Could you share how you activated your nodes?

ckt1010, 2017-03-28:
Hi @brocaar , thanks for your reply!
I first power on node A which is "4786e6ed00380057". After join, every thing is fine, and data parse ok.
Then I power on node B which also use "4786e6ed00380057". It will trigger server accpect node B and I can see device address, network session key, application session key has been changed.
I thought packet from node A will reject then. But actually not.... It still get ACK, and server still decrypt packet, but use new application session key, so it's wrong. Because MIC verify and ACK is ok, I think device address and network session key is still use old for node A.

brocaar, 2017-03-28:
Could you share which version of LoRa Server you're using? In the meanwhile, I'll see if I can reproduce this behaviour.

brocaar, 2017-03-28:
I'm not able to reproduce your issue.

What I tried:

1. Added node `0101010101010101` to LoRa App Server
2. Provisioned two nodes (node1, node2) with the same DevEUI / AppEUI / AppKey as for node `0101010101010101`
3. Activated node1 using OTAA
4. Activated node2 using OTAA
5. Now when node1 sends data, I get the (expected) error `ERRO[0755] processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic  data_base64="QGEsFgeABAACuXLiq4BlZ+hKprcHrOsfSz2p78w="`

ckt1010, 2017-03-29:
Hi @brocaar ,
Would you please tell me how long you wait between step 3 and step 4 ? For me, I wait about one minute.

ckt1010, 2017-03-29:
Sorry @brocaar , it maybe code version's problem. After I update my code, this problem disappear.
But another problem appear... Please check log below:

> application/1/node/4786e6ed00380057/join {"applicationID":"1","applicationName":"70b3d57ed000140d","nodeName":"4786e6ed00380057","devEUI":"4786e6ed00380057","devAddr":"07662e52"}
gateway/b827ebfffe6c9ff7/tx {"txInfo":{"mac":"b827ebfffe6c9ff7","immediately":false,"timestamp":3675438499,"frequency":475500000,"power":14,"dataRate":{"modulation":"LORA","spreadFactor":7,"bandwidth":125},"codeRate":"4/5","iPol":null},"phyPayload":"IIiloMTT95ewJxMRrsPC8dURtz33t8Abemq5BRBRurEU"}
application/1/node/4786e6ed00380057/rx {"applicationID":"1","applicationName":"70b3d57ed000140d","nodeName":"4786e6ed00380057","devEUI":"4786e6ed00380057","rxInfo":[{"mac":"b827ebfffe6c9ff7","rssi":-95,"loRaSNR":3.2}],"txInfo":{"frequency":475700000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":true,"codeRate":"4/5"},"fCnt":3,"fPort":2,"data":"xStAvgOiEMo7kvg4"}

Decryption of old node data still have problem. But it happened only once and at new node join. Test step same as before.
Now I give my suspect below:
1. node1 send new data to server.
2. node1 data pass loraserver's mic verify.
3. node2 join and session renewed.
4. node1 data send to lora-app-server
5. lora-app-server decryption it with new App session key.
6. bad data appear....
So the result is loraserver and lora-app-server has different session at some critical region.
Do you think it is make sense? Thanks!

brocaar, 2017-03-29:
Could you define "bad data"? Note that `xStAvgOiEMo7kvg4` is already decrypted, you only have to base64 decode it (e.g. https://play.golang.org/p/_1sEX3j8xX).

ckt1010, 2017-03-30:
Sorry for confusion.
"Bad data" means use wrong Application session key to decrypt payload. It will cause sensor data seems abnormal, like -105℃.....

brocaar, 2017-03-30:
It seems that I can't replicate your issue. When node1 sends data again after node2 joins (with the same deveui), I get an (expected) error in the LoRa Server logs `ERRO[0755] processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic data_base64="QGEsFgeABAACuXLiq4BlZ+hKprcHrOsfSz2p78w="`.

That error is expected since the node-session for the deveui has been overwritten by the activation of node2. So the MIC of node1 is no longer valid because the NwkSKey changed.

ckt1010, 2017-03-31:
Hi @brocaar ,
As I said it is a critical region problem, I reproduce it only once.
Maybe you should check if my suspect is available at first.

brocaar, 2017-03-31:
Theoretically what could have happened (as the MIC is validated in LoRa Server) and the payload is decrypted in LoRa App Server is that the activation of the second node wasn't fully processed yet. Thus the `AppSKey` was written to the PostgreSQL db in LoRa App Server, but LoRa App Server didn't reply yet to LoRa Server with the join-accept. Because of this, there was a small time in which LoRa Server still accepted data from "node1".

If that was the case, I'm not sure if anything can be done about it, as we're dealing with two data-stores that each do their separate commit.

ckt1010, 2017-03-31:
Hi @brocaar ,
Maybe we can tag every packet between LoRa App Server and LoRa Server with version number. Once node join accept, LoRa App Server will reject this EUI's packet which version is older than join request packet.

brocaar, 2017-04-04:
Don't think that will make it any cleaner. Honestly, I don't think it is an issue, as long there is no real security issue. In this case both nodes share both the `DevEUI` and the `AppKey` (else you would not be able to activate the second node). Also LoRa Server doesn't support multiple nodes using the same DevEUI (as the DevEUI is used as unique ID).

ckt1010, 2017-04-11:
En... I just post some suggestion. Maybe it's not critical like you say.

closed 2017-06-21 by brocaar