brocaar/chirpstack-network-server #149: Ping database error after upgrading to loraserver 0.16.0

maximilianschwab, 2017-03-28:
Hi guys,

after upgrading to loraserver version 0.16.0 i get a ping database error. The database is up and running, i havent changed the db password. The OS is Linux 16.04.02 LTS. Everything was fine before upgrading to the newest loraserver version.

`root@v22016103605938256:/etc/default# journalctl -u loraserver -f -n 50
-- Logs begin at Di 2017-03-28 16:40:50 CEST. --
Mär 28 16:48:21 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:21+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
Mär 28 16:48:23 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:23+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
Mär 28 16:48:25 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:25+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
Mär 28 16:48:27 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:27+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
Mär 28 16:48:29 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:29+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
Mär 28 16:48:31 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:31+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
Mär 28 16:48:33 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:33+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
Mär 28 16:48:35 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:35+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
Mär 28 16:48:37 v22016103605938256 loraserver[1837]: time="2017-03-28T16:48:37+02:00" level=error msg="ping database error, will retry in 2s: pq: password authentication failed for user "loraserver""
`

Are there any suggestions on how to solve this error?

Kind regards,
Max

brocaar, 2017-03-29:
Note that you need to configure also a database for LoRa Server (before this was only for LoRa App Server). See also this note: https://github.com/brocaar/loraserver/releases/tag/0.16.0

maximilianschwab, 2017-03-29:
Sorry! This solved the problem!
Thank you!

closed 2017-03-29 by maximilianschwab

vikkymahdrey, 2018-05-28:
I installed loraserver 0.26.3 and I configured postgresql 9.5.12 but continuously getting ping database error.
In loraserver configuration file, I added POSTGRES_DSN="postgres://loraserver_ns:loraserver_ns@localhost/loraserver_ns?sslmode=disable". I checked username password and database are correct.
Could you please help us