brocaar/chirpstack-network-server #150: Node-session does not exist or invalid fcnt or mic

RCOdev, 2017-03-28:
Hi,
I've installed the lora server (lora-gateway-server, loraserver, lora-app-server) [really good job by the way].
I've got a Lorank8 gateway and ATIM node.

I'm using ABP mode, the node is activated (seen in lora-app-server log) but loraserver give me this error :
msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic"

Do you have an idea of why I get this error?

Thanks!

brocaar, 2017-03-29:
There are a couple things that could go wrong:

* Node-session does not exist
* Node-session exists, but with incorrect keys (note that for some nodes you have to enter them LSB, for some MSB thus `0102030405060708` might in fact be `0807060504030201`)
* The FCnt is invalid (compare the frame-counters on your node with LoRa App Server)

RCOdev, 2017-03-29:
Thanks for your comment.

Node uplink counter is 00000001 and downlink counter is 00000000.
In LoRa App Server it is set to 1 (uplink) and 0 (downlink).

All my nodes IDs are LSB first. So for DevAddr = 1A0200A0, I wrote A000021A in Activation page in LoRa App Server.

Deepdone, 2017-04-04:
i have the same error msg,but my node had join the net and the server can receive msg from the node.Is it because there is a error data when the server works well , lead to have this error message?then the server can receive the right data from the node ,but can not handle it?@

mikumaycry, 2017-04-10:
I got the same error.
It always occurred when gateway send two packet from the same node at one time.
As a result, the first request is handled but the response is failed to publish to gateway.  
I think there's some problem in handling duplicated packets.

dhinojosac, 2017-05-23:
Hi, i have the same error. This happens when i reset the node and set the paramaters again, but the FCount in loraserver still has the older values.  Is there any way for the server to reset the counter??

brocaar, 2017-05-24:
Reactivate the node (in LoRa App Server) with relax frame-counter mode. Please note that this will introduce a security risk! Better is to use OTAA activation after every node reset / power-cycle.

dhinojosac, 2017-05-24:
Thank you @brocaar . We resolve it doing a re-activation post request at moment that we reset the node (without relax frame-counter mode).

closed 2017-06-21 by brocaar

gousmodin, 2017-10-28:
I have enabled relax frame counter, and reactivated the node. After restart i  getting uplink data but no doownlink. I have set node for confirmed msg.