brocaar/chirpstack-network-server #151: Using stored procedure to remove SQL handling in GO

pmiam999, 2017-03-29:
Hi Brocaar,

Looking at your latest commits regarding gateway and timezone management, i was wondering why you've choosen to perform raw SQL queries in your GO code and not using pg stored procedure ?

brocaar, 2017-03-29:
I haven't considered using stored procedures to be honest. What would be the benefit?

(I think it wouldn't have solved the mistake that I made regarding timezones. The mistake that I made was that the db was possibly operating in a different tz than LoRa Server. Therefore the `generate_series` was not taking DST changes between a start and end timestamp into account.)

pmiam999, 2017-03-29:
Yes i've seen your mistake, pretty fast detection and fix. Congrats.

To my experiences, the main benefit using stored procedures are :
- running business logical in transaction is simpler, keeping SQL overhead away from your code (begin / commit / rollback nightmare)
- you can perform sanity/pre-flight checks in one place, using advisory lock to enforce your database consistency
- you can write your stored procedure in great languages such as plv8 (in case you have to deal with json data, that's really powerfull)
- you just have to reload your store procedures to upgrade without touching any part of the code (as long as the prototype arguments are identicals)

closed 2017-03-30 by pmiam999