brocaar/chirpstack-network-server #155: "permission denied for relation gorp_migrations" on Raspberry Pi 3 mod B

Yaoshicn, 2017-04-03:
Hi, brocaar,

I am working with Raspberry Pi 3 mod B (Raspbian Jessie). I followed the tutorial and finished the setup. However, I got **"applying migrations failed: pq: permission denied for relation gorp_migrations"** error when I started the lora server. I have granted the privileges of database table to user loraserver_as and loraserver_ns.

Would you please give some help on this issue? Thanks in advance.

Best Regards,
Xuanliang

brocaar, 2017-04-03:
To me it seems that your `POSTGRES_DSN` / `--postgres-dsn` setting might be incorrect / does not contain the password that you set for your role.

Yaoshicn, 2017-04-03:
Thank you for the correction. Now I can start the server without errors. But the web interface shows "Error pq: relation "user" does not exist (code: 2)".

My configs go like this:
Server:
```
loraserver --net-id "010203" --band "EU_863_870" --postgres-dsn "postgres://loraserver_ns:dbpassword@localhost/loraserver_ns?sslmode=disable" --db-automigrate
```
App server:
```
lora-app-server --postgres-dsn "postgres://loraserver_as:dbpassword@localhost/loraserver_as?sslmode=disable" --jwt-secret "C1TXibQJsfquG9C6sPhks6h6zDwMrIlQYv8sDzlaYGs=" --log-level="5" --http-tls-cert="/home/pi/certs/http.pem" --http-tls-key="/home/pi/certs/http-key.pem"
```

The password for user loraserver_as and loraserver_ns is "dbpassword" which is default value in the doc.

Thank you for your help and patience.

brocaar, 2017-04-03:
@Yaoshicn add `--db-automigrate` to the `lora-app-server` command and all should be fixed :-)

closed 2017-04-03 by brocaar

Yaoshicn, 2017-04-03:
It works! Thank you!