brocaar/chirpstack-network-server #156: I have a question about time stamp value of Join Accept

thisdothyuu, 2017-04-03:
Hello brocaar.

Your source code has been a great help to my research. I really thank you for your hard work.

Today, I have a question about JoinAceptDelay.

In JoinAccept.go file, getJoinAcceptTxInfo() function set the TimeStamp value of the join accept packet by following sentence. 

`txInfo.Timestamp = rxInfo.Timestamp + uint32(common.Band.JoinAcceptDelay1/time.Microsecond)`

My question is, why JoinAcceptDelay value is added? 

As I understand, after the Join Request is sent, RX1 window is open and it is closed after the pre-defined delay. And I confirmed that [LoRaMAC packet forwader](https://github.com/Lora-net/packet_forwarder) sends packet at the timestamp assigned by the Network Server. Thus, it looks like the packet will arrive after rx1window is closed by adding JoinAcceptDelay. However, it works well. 

And I also confirmed that, when I changed the TimeStamp value, JoinAccpet is not received by end node. 

I couldn't understand how this happens. JoinAcceptDelay is added by Network Server and RX1Window is opened during the JoinAcceptDelay. How packet is arrived in time on end node?? What am I missing?

brocaar, 2017-04-03:
Great to hear!

RX1 for the join-accept receive window is 5 seconds after the join-request has been transmitted. The `rxInfo.Timestamp` holds the (gateway) timestamp when the request was received, thus to match RX1, it needs to be incremented by 5 seconds (which is the value of `JoinAcceptDelay1`).

Note that RX1 **opens** after 5 seconds, it does not stay open until RX1. So the join-accept has to be precisely timed.

thisdothyuu, 2017-04-03:
Thank you sooooooooooooooooooooooo much.

I have misread the document. 

Thank you again.

closed 2017-04-03 by thisdothyuu