brocaar/chirpstack-network-server #157: Failed to start LoRa App Server.

VahrushevSA, 2017-04-07:
After version 0.5.0 I can not run Lora app server. I ask to help, what can be done.

brocaar, 2017-04-07:
Duplicate of https://github.com/brocaar/lora-app-server/issues/63

closed 2017-04-07 by brocaar

vikkymahdrey, 2017-07-02:
[image]

vikkymahdrey, 2017-07-02:
please help on this

vikkymahdrey, 2017-07-02:
I don't know exactly that i'm going correct way or not for generating jwt.
Please check here
[image]

vikkymahdrey, 2017-07-02:
i'm taking JWT_SECRET=uAkvtgcPuaNxRubHA0g+OlS8ODFMkqJHRj+N700zzQY
This is the generated one.
Is this correct or not?

vikkymahdrey, 2017-07-02:
M using Ubuntu..
Now I configured loraserver and loraappserver
here m getting
[image]

But m not able to open loar-app-server web interface on http://localhost:8089
I configure Http_bind at localhost:8089 in /etc/default/lora-app-server
please check and help me

sagar-patel-sls, 2018-10-05:
1. start Lora-App-Server 
sudo /etc/init.d/lora-app-server [start|stop|restart|status]
OR
sudo systemctl [start|stop|restart|status] lora-app-server

2. access web interface using https://localhost:8089