brocaar/chirpstack-network-server #159: selecting correct packet from gateway

uncle-evil, 2017-04-10:
i am using gateways(semtech packet forwarder) working at CN470-510 band, some times a node send one packet, the gateway will receive two or more packets of that node at the same time, **this packets are different in frequency, RSSI and SNR, very close timestamp(always less than 10uS), but same payload, same bindwidth, same SF, CRC OK.**

i found out that the LoRa server always take the first arrived packet to handle node join in request and comfirmed data up.
for example, if a node(using RX1) sending a join in request or comfirmed data up at 472.9MHz, the gateway received 2 packets at 472.7MHz and 472.9MHz, the server will send join in accept or ACK data at 502.7MHz, but the node was expect a packet on RX1 at 502.9MHz.

**one solution is that the gateway drop the packets that has a pool RSSI and SNR, but i think it's better to be handled by LoRa server.**

packet log of the gateway:
```json
{"rxpk":[{"tmst":591655212,"chan":2,"rfch":1,"freq":472.700000,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":-9.5,"rssi":-91,"size":17,"data":"QNegBQaAMgAII0nuKL2t1hQ="},{"tmst":591655204,"chan":3,"rfch":1,"freq":472.900000,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":9.8,"rssi":-46,"size":17,"data":"QNegBQaAMgAII0nuKL2t1hQ="}]}
```

brocaar, 2017-04-10:
I'm not familiar with nodes sending the same frame over multiple frequencies at the same time, I'm not sure if this is correct behaviour...

What LoRa Server does it collects all frames (during a given period of time, e.g. 200-300ms), sorts the frames received by the gateways on SNR and then uses the gateway with the best SNR. See https://github.com/brocaar/loraserver/blob/master/internal/uplink/collect.go#L25.

brocaar, 2017-04-10:
I just checked with @tftelkamp, probably the issue is that your node is really close to your gateway. Because of the sensitivity of your gateway, it will receive the frame over two channels.

uncle-evil, 2017-04-10:
i test it in door, node is about 3 ~ 5M away from gateway. i will check it out door soon.

uncle-evil, 2017-04-11:
I have tested this issue again, if the node is far from gateway (about RSSI < -45dBm), it seens duplicated packets over two channels disappeared. @brocaar thanks!

But still i suggest the server handles such issue just in case.

mikumaycry, 2017-04-12:
I could repeat this case when I set node's antenna straight and gateway's antenna horizontal.
The packet resent count is set to 2 on node.
So node would sent 2 same packet on different frequency.
Server always handle the first one and drop the latter, ignoring RSSI or SNR.
But node expects the latter packet's receive frequency, so node could never receive data.
here is part of the logs on gateway:

JSON up: {"rxpk":[{"tmst":1925349980,"chan":3,"rfch":1,"freq":472.900000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":8.8,"rssi":-20,"size":33,"data":"gCtW8waASQAIh4xBnpNBiOva/TUIIp9sFgf8I8VyTFpj"},{"tmst":1925349980,"chan":7,"rfch":0,"freq":472.100000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-17.2,"rssi":-87,"size":33,"data":"gCtW8waASQAIh4xBnpNBiOva/TUIIp9sFgf8I8VyTFpj"}]}
INFO: [up] PUSH_ACK received in 0 ms

JSON up: {"rxpk":[{"tmst":1931351220,"chan":3,"rfch":1,"freq":472.900000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":6.2,"rssi":-21,"size":33,"data":"gCtW8waASQAIh4xBnpNBiOva/TUIIp9sFgf8I8VyTFpj"}]}

JSON up: {"rxpk":[{"tmst":1920350772,"chan":5,"rfch":0,"freq":471.700000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-18.0,"rssi":-87,"size":33,"data":"gCtW8waASQAIh4xBnpNBiOva/TUIIp9sFgf8I8VyTFpj"},{"tmst":1920350772,"chan":1,"rfch":1,"freq":472.500000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":8.5,"rssi":-20,"size":33,"data":"gCtW8waASQAIh4xBnpNBiOva/TUIIp9sFgf8I8VyTFpj"}]}

JSON down: {"txpk":{"imme":false,"tmst":1921350772,"freq":501.7,"rfch":0,"powe":14,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":26,"data":"YCtW8wYgSAAIRwKeEIuJontcLM3uvmYhgxs="}}

brocaar, 2018-04-15:
This will be fixed by LoRaWAN 1.1 as the channel-number is part of the MIC in 1.1. This makes it possible for the network-server to determine the correct TX channel number :-)

closed 2018-04-15 by brocaar