brocaar/chirpstack-network-server #160: Is loraserver implemented MAC command?

Jungwoon-Lee, 2017-04-21:
Hello, brocaar. Thank you for making great project.

I have been researched your lora project and LoRaWAN.

I builded loraserver, lora-app-serever, G/W bridge and etc on windows 10 environment also I analyzed your code on the loraserver. 

I checked that the loraserver can recieve packet of type of MAC Command. However I could not find the MAC-command to control End-device and part generating the MAC command from LoRa network server.

Where is that function which process and generate MAC command  implemented ??

I checked the code that when loraserver recieve the MAC command, the loraserver verify packet and send to nc using grpc.

I guess that you are going to develope network-controller.  processing and generating of MAC command will be implemented on network-controller??

I wait for your reply. Thank you.

brocaar, 2017-04-25:
Initially I was planning to offer a network-controller, however most MAC commands are really tight coupled to the network-server (as they control data-rates, frequencies, etc..). Therefore it would be better to to first know what you're planning to do with the MAC commands :-)

dietb, 2017-04-26:
I think this would be a good idea, too.  An easy use case would be something like changing spreading factors manually, like "force a device that you _know_ wont be near a gateway to SF10."

I'd also like to see what MAC commands the network server is sending down, but those don't appear to be stored in MQTT currently.

brocaar, 2017-05-18:
Please note that LoRa Server implements a gRPC API and has a method to schedule downlink mac-commands: https://github.com/brocaar/loraserver/blob/master/api/ns/ns.proto#L25

However, this might some re-work to add support for sending mac-commands blocks that must be transmitted in a single PHYPayload.

I'm also wondering if a single API method to send all mac-commands or having an API call per mac-command would be better. E.g. you would call the:

`LinkADRReq` API method, LoRa Server would then call the `LinkADRAns` when the node answers your`LinkADRReq` request.

brocaar, 2017-07-10:
I forgot to update this issue, but sending mac-commands through an api has been implemented in 0.19.0 (see https://github.com/brocaar/loraserver/releases).

closed 2017-07-10 by brocaar

carlberg74, 2017-11-18:
Could you maybe give and example how to send mac-commands, what API (nodeDownlink POST?) and what to send for setting ex. SF.