brocaar/chirpstack-network-server #161: Network Quality Question

Chill0309, 2017-04-23:
Hi, should running a gateway as a packet forwarder and having the network server in the cloud run ok if the gateway is only connected with a 3G/GPRS SIM connection rather than a 'good', wired internet connection? Thanks.

brocaar, 2017-04-25:
The think you need to take care about is the roundtrip time of packets between the "cloud" and your gateway. For OTAA you have max 5 second, for downlink data by default 1 second. Note the latter one is configurable (Receive window delay).

Could you close this issue when it answers your question?

Chill0309, 2017-04-25:
Thanks, I think this fits ok with the type of pings we're getting. Do you have any gauge for how capable it is dealing with heavy loads on the lora side, i presume its all quite lightweight and scales fairly well?

brocaar, 2017-09-04:
Lets move this topic to https://forum.loraserver.io/ in case you have any further questions :-)

closed 2017-09-04 by brocaar