brocaar/chirpstack-network-server #162: PostgreSQL 9.5+ not available in compiled version for RPi, but required for LORASERVER

glederer007, 2017-04-26:
Hello Orne,

In the documentation you mention that PostgreSQL 9.5+ version is required for LORASERVER. As far as I can see, PostgreSQL for ARM processors is not available from www.postgresql.org. This means that it is not possible to download it in the way you wrote in the instructions. 
The latest version of PostgreSQL for RPi is 9.4.10 (available through APT-GET). Installing PostgreSQL to a RPi by locally compiling it is not an easy task. It can be done, but it is not easy. Is PostgreSQL 9.5+ version really needed, or it will work with 9.4.10 as well?

Regards,

Gábor

brocaar, 2017-04-26:
Yes, 9.5 is required. However it seems a 9.6 version is already in Raspbian:
http://archive.raspbian.org/raspbian/pool/main/p/postgresql-9.6/

Alternatively, you could use the Jessie backports repo:
https://backports.debian.org/Instructions/

See also: https://github.com/brocaar/loraserver-pi-gen for a script to generate Raspberry Pi images :-)

glederer007, 2017-04-26:
Orne,

Ok, thanks.

Gabor,

PS: Incredible how you handle this, you have a solution for everything...

closed 2017-04-26 by glederer007