brocaar/chirpstack-network-server #163: Build source

kruit, 2017-04-26:
Hello Orne, hello guys,

Trying to build the loraserver with make as I'm not familiar with docker. The compiler stumbles on the first line with `go-generate` command in main.go with a message: `cmd/loraserver/main.go:1: running "go-bindata": exec: "go-bindata": executable file not found in $PATH`. 
It's a clean project just after git clone, the bin directory is empty. Compilation without the `go-generate` instruction claims as expected about a non-existent 'loraserver/internal/migrations' package...
Please, could you point me the right way to compile?

utopiaprince, 2017-04-28:
please install go-bindata.
```
go get -u -v github.com/jteeuwen/go-bindata
```
then go to $GOPATH. do this:
```
go install github.com/jteeuwen/go-bindata
```

kruit, 2017-04-28:
so easy, thank you a lot!

closed 2017-04-28 by kruit

OdysLam, 2019-07-24:
I am trying to do it inside a docker container but it doesn't seem to work 
from Dockerfile:
```
WORKDIR $PATH
RUN go get -u -v github.com/jteeuwen/go-bindata
RUN go install github.com/jteeuwen/go-bindata
```