brocaar/chirpstack-network-server #166: new build loraserver Regional Parameters RU864_869

VahrushevSA, 2017-04-29:
Hello Brocaar!
I ask to compile a new build loraserver on the basis of regional parameters RU864_869
Frequency:
864.1
864.3
864.5
RX2Frequency:
869.525
[Частотный план RU868 ENG.pdf](https://github.com/brocaar/loraserver/files/965959/RU868.ENG.pdf)

brocaar, 2017-05-04:
Could you clarify on which data these values are bases? I remember there was a LoRa Alliance draft for RU a while ago, however this seems to be a custom specification. I would rather stick to the official draft, but I'm a bit confused what to take as the actual truth (as I don't want to break config for other uses).

In case you want to customize for your own needs, you could modify this file https://github.com/brocaar/loraserver/blob/master/vendor/github.com/brocaar/lorawan/band/band_ru864_869.go, and then recompile as described in the `README.md`.

VahrushevSA, 2017-05-08:
Thank you very much Brocaar!,
I managed to compile the loraserver for my needs.

closed 2017-05-09 by brocaar