brocaar/chirpstack-network-server #168: make build with undefined migrations.AssetDir error

VahrushevSA, 2017-04-30:
Compiling source for linux amd64
# command-line-arguments
cmd/loraserver/main.go:114: undefined: migrations.AssetDir
Makefile: 8: failed to complete the recipe for the "build" target
Make: *** [build] Error 2

how to fix this? thanks

utopiaprince, 2017-05-03:
please do this first：
```
make statics
```
then you will get internal/migrations/migrations_gen.go file.
after that, try:
```
make build
```

dreamerzjy, 2017-05-04:
there is no AssetDir in file"migrations_gen.go"

root@loraserver:~/loraserver/bin# go-bindata -version
go-bindata 3.1.0 (Go runtime go1.2).
Copyright (c) 2010-2013, Jim Teeuwen

root@loraserver:~/loraserver/bin# go-bindata-assetfs -version
go-bindata 3.1.0 (Go runtime go1.2).
Copyright (c) 2010-2013, Jim Teeuwen.
Cannot read bindata.go open bindata.go: no such file or directory

brocaar, 2017-05-04:
Are you using the latest version from the `master` branch?

Could you try with the provided `docker-compose` environment? The `Dockerfile` will install all the required dependencies to test and compile.

To start a shell `docker-compose run --rm loraserver bash`

Then inside that shell try again `make build`.

dreamerzjy, 2017-05-04:
sorry,i found the reason is i installed "go-bindata:"by  use"apt-get install go-bindata";
after i use "apt-get autoremove go-bindata",problem is fixed.

closed 2017-05-04 by brocaar