brocaar/chirpstack-network-server #170: Class C network: Getting "maximum payload size exceeded" when payload > 51bytes on downlink

reixd, 2017-05-02:
Since I updated to the latest release of loraserver I am getting the following error message (displayed in lora-app-server but it comes from the loraserver stack) when sending a downlink message to a Class C mote.

> ERRO[1575] handle data-down payload error: push data-down error: rpc error: code = 3 desc = maximum payload size exceeded  application_id=1 dev_eui=313430365e338813 reference=ba9010e5-bc12-4a32-b31c-0e058641c5a8

- I am getting this error when my application sends data bigger than 51bytes.
- My application detects the SF of the mote and decides which is the max payload allowed for the datarate based on the LoRa specification (51B, 115B and 222B). After the loraserver detects over ADR that the mote can uses better SF than 12,11&10 the mote stops receiving downlink messages. 
- Uplink messages are still working.
- Downlink messages (send at the same time) with payload length <=51bytes reach the mote too.

I suspect that the loraserver is not updating the node session correctly with the new SF and DR. But I was unable to find the code, without digging too deep.

In addition the testsuite/class_c_test.go is not testing all edge cases for max payload.
The test case suite needs to be extended to test against all SF and DR and the respective payload max lenghts.

reixd, 2017-05-02:
I just fund the solution: **Configure the RX2 datarate to 6**

Did not have in mind, that the DL DR is fixed. And because I choose a high DR the application did not collide with the different SF settings.

### lora-app config

network settings:
- Class C: True
- ABP: False  --> OTAA active
- RX1: True
- Enable relax frame-counter: True  (this is a test app, yes I am aware of the security implications)
- Receive window delay: 0
- RX1 DR offset: 0
- **RX2 DR: 6**
- Channel List: EU 863-870 [867100, 867300, 867500, 867700, 867900]
- ADR interval: 3 (low number to change SF/DR quickly)
- dB margin: 2 (devices are on the same lab)

brocaar, 2017-05-04:
Great your issue has been solved.

> In addition the testsuite/class_c_test.go is not testing all edge cases for max payload.
The test case suite needs to be extended to test against all SF and DR and the respective payload max lenghts.

Please note that there is a test that the max allowed payload is enforced. I'm not sure if there should be a test for every SF / DR combination. These values are derived from the [lorawan/band](https://godoc.org/github.com/brocaar/lorawan/band) package and vary per region (and per configuration, e.g. the [AS923](https://github.com/brocaar/lorawan/blob/master/band/band_as923.go#L13) numbers depend on dwell-time and repeater compatibility). That would be a lot of tests ;-) Therefore I think it is good to test that the logic (lookup) is in place, but do not worry about all the possible values.

reixd, 2017-05-04:
> Therefore I think it is good to test that the logic (lookup) is in place, but do not worry about all the possible values.

seems legit ;)

closed 2017-05-04 by reixd