brocaar/chirpstack-network-server #171: pq: syntax error at or near "on"

JasonZhang91503, 2017-05-08:
Hi Brocaar
I've install your server(lora-gateway-bridge, loraserver, lora-app-server)
and all installed successfully

I create a node and set devaddr,nwskey,appskey and abp mode
I test this node by sending some data

but i've some error log :
time="2017-05-08T21:01:40+08:00" level=error msg="handle stats packet error: aggregate gateway stats error: insert or update aggregate error: pq: syntax error at or near "on"" 

This log sometimes appears,
but I have no idea what problem it is.
Could you help me?

brocaar, 2017-05-08:
Did you create a separate database for LoRa Server and LoRa App Server? It is probably that, or you forgot to run the database migrations (see `--db-automigrate` / `DB_AUTOMIGRATE` config flag).

brocaar, 2017-05-12:
Also, make sure you installed PostgreSQL 9.5+

JasonZhang91503, 2017-06-12:
Hi Brocaar
I am very sorry for not having to reply to you for a long time
because I am busy with my studies.

I check that I separate database for LoRa Server and LoRa App Server.
And I'm try to upgrade loraserver, lora-app-server component to last version,
the error log(pq: syntax error at or near "on")  no longer appears
But there another log say "type jsonb does not exist" when loraserver try to migration.
I try to install PostgreSQL 9.6 , but not work.
And I find that my system has both PostgreSQL 9.6 and PostgreSQL 9.3.
So I uninstall both of them and install PostgreSQL 9.6 again.
There is no error log!

Thank you brocaar, I resolve my problem!

closed 2017-06-19 by brocaar

alexanderplanar, 2017-10-16:
Hi, guys! I just faced to a similar trouble, but the solution was instructive. 
When I had been configuring the postgresql I missed the fact the version must be 9.5+, and then I installed 9.6 without removing old version 9.4. But apparently the new version use the neighbour port, If the old version is still working. I thinked I solve the problem, but the application have been use 9.4 version all the time instead 9.6, and I did'nt know about it. When I faced to this problem (syntax error), I decided to check the version is used now using netstat utility. Pay attention, if anyone faced to this problem, do not forget to delete the packets (postgresql) with configuration files! Otherwise, the newly installed version (e.g. 9.6) will  not use 5432 port (in my case it was 5433).