brocaar/chirpstack-network-server #174: loraserver version=0.16.1-2-gdde77eb issue

Richaagarwalrn, 2017-05-19:
Hi,
Loraserver get stopped and started repeatedly.
May 18 19:51:34 pc2 loraserver[6659]:#  panic: value method gateway.GPSPoint.Value called using nil *GPSPoint pointer
May 18 19:51:34 pc2 systemd[1]: loraserver.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
May 18 19:51:34 pc2 systemd[1]: loraserver.service: Unit entered failed state.
May 18 19:51:34 pc2 systemd[1]: loraserver.service: Failed with result 'exit-code'.
May 18 19:51:34 pc2 systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
May 18 19:51:34 pc2 systemd[1]: Stopped LoRa Server.
May 18 19:51:34 pc2 systemd[1]: Started LoRa Server.

**Log** 
May 18 19:50:03 pc2 systemd[1]: Stopped LoRa Server.
May 18 19:50:03 pc2 systemd[1]: Started LoRa Server.
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.16.1-2-gdde77eb
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="setup redis connection pool" url="redis://localhost:6379"
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="connecting to postgresql"
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="connecting to application-server" ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="backend/gateway: connected to mqtt server"
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="backend/gateway: subscribing to stats topic" topic="gateway/+/stats"
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="no network-controller configured"
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="migrated node-sessions to new format" errors=0 migrated=0
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="applying database migrations"
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="migrations applied" count=0
May 18 19:50:03 pc2 loraserver[6627]: time="2017-05-18T19:50:03+05:30" level=info msg="starting api server" bind="0.0.0.0:8000" ca-cert= tls-cert= tls-key=
May 18 19:50:09 pc2 loraserver[6627]: time="2017-05-18T19:50:09+05:30" level=info msg="backend/gateway: rx packet received"
May 18 19:50:09 pc2 loraserver[6627]: time="2017-05-18T19:50:09+05:30" level=info msg="packet(s) collected" dev_eui=2222222222222222 gw_count=1 gw_macs=aa555a0000000101 mtype=ConfirmedDataUp
May 18 19:50:09 pc2 loraserver[6627]: time="2017-05-18T19:50:09+05:30" level=info msg="rx info sent to network-controller" dev_eui=2222222222222222
May 18 19:50:09 pc2 loraserver[6627]: time="2017-05-18T19:50:09+05:30" level=warning msg="get gateways for macs error: expected 1 gateways, got 0" macs=[aa555a0000000101]
May 18 19:50:09 pc2 loraserver[6627]: time="2017-05-18T19:50:09+05:30" level=info msg="node-session saved" dev_addr=22222222 dev_eui=2222222222222222
May 18 19:50:09 pc2 loraserver[6627]: time="2017-05-18T19:50:09+05:30" level=info msg="node-session saved" dev_addr=22222222 dev_eui=2222222222222222
May 18 19:50:10 pc2 loraserver[6627]: time="2017-05-18T19:50:10+05:30" level=info msg="received data down from application" confirmed=true data_base64=bn0k dev_eui=2222222222222222 fcnt=46 more_data=true
May 18 19:50:10 pc2 loraserver[6627]: time="2017-05-18T19:50:10+05:30" level=info msg="backend/gateway: publishing tx packet" topic="gateway/aa555a0000000101/tx"
May 18 19:50:17 pc2 loraserver[6627]: time="2017-05-18T19:50:17+05:30" level=info msg="backend/gateway: rx packet received"
May 18 19:50:17 pc2 loraserver[6627]: time="2017-05-18T19:50:17+05:30" level=info msg="packet(s) collected" dev_eui=2222222222222222 gw_count=1 gw_macs=aa555a0000000101 mtype=ConfirmedDataUp
May 18 19:50:17 pc2 loraserver[6627]: time="2017-05-18T19:50:17+05:30" level=info msg="rx info sent to network-controller" dev_eui=2222222222222222
May 18 19:50:17 pc2 loraserver[6627]: time="2017-05-18T19:50:17+05:30" level=warning msg="get gateways for macs error: expected 1 gateways, got 0" macs=[aa555a0000000101]
May 18 19:50:17 pc2 loraserver[6627]: time="2017-05-18T19:50:17+05:30" level=info msg="node-session saved" dev_addr=22222222 dev_eui=2222222222222222
May 18 19:50:18 pc2 loraserver[6627]: time="2017-05-18T19:50:18+05:30" level=info msg="node-session saved" dev_addr=22222222 dev_eui=2222222222222222
May 18 19:50:18 pc2 loraserver[6627]: time="2017-05-18T19:50:18+05:30" level=info msg="received data down from application" confirmed=true data_base64="//Hu" dev_eui=2222222222222222 fcnt=47 more_data=true
May 18 19:50:18 pc2 loraserver[6627]: time="2017-05-18T19:50:18+05:30" level=info msg="backend/gateway: publishing tx packet" topic="gateway/aa555a0000000101/tx"
May 18 19:50:25 pc2 loraserver[6627]: time="2017-05-18T19:50:25+05:30" level=info msg="backend/gateway: rx packet received"
May 18 19:50:26 pc2 loraserver[6627]: time="2017-05-18T19:50:26+05:30" level=info msg="packet(s) collected" dev_eui=2222222222222222 gw_count=1 gw_macs=aa555a0000000101 mtype=ConfirmedDataUp
May 18 19:50:26 pc2 loraserver[6627]: time="2017-05-18T19:50:26+05:30" level=info msg="rx info sent to network-controller" dev_eui=2222222222222222
May 18 19:50:26 pc2 loraserver[6627]: time="2017-05-18T19:50:26+05:30" level=warning msg="get gateways for macs error: expected 1 gateways, got 0" macs=[aa555a0000000101]
May 18 19:50:26 pc2 loraserver[6627]: time="2017-05-18T19:50:26+05:30" level=info msg="node-session saved" dev_addr=22222222 dev_eui=2222222222222222
May 18 19:50:26 pc2 loraserver[6627]: time="2017-05-18T19:50:26+05:30" level=info msg="node-session saved" dev_addr=22222222 dev_eui=2222222222222222
May 18 19:50:26 pc2 loraserver[6627]: time="2017-05-18T19:50:26+05:30" level=info msg="received data down from application" confirmed=true data_base64=s6F1 dev_eui=2222222222222222 fcnt=48 more_data=true
May 18 19:50:26 pc2 loraserver[6627]: time="2017-05-18T19:50:26+05:30" level=info msg="backend/gateway: publishing tx packet" topic="gateway/aa555a0000000101/tx"
May 18 19:50:33 pc2 loraserver[6627]: time="2017-05-18T19:50:33+05:30" level=info msg="backend/gateway: gateway stats packet received" mac=aa555a0000000101
May 18 19:50:33 pc2 loraserver[6627]: panic: value method gateway.GPSPoint.Value called using nil *GPSPoint pointer
May 18 19:50:33 pc2 systemd[1]: loraserver.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
May 18 19:50:33 pc2 systemd[1]: loraserver.service: Unit entered failed state.
May 18 19:50:33 pc2 systemd[1]: loraserver.service: Failed with result 'exit-code'.
May 18 19:50:33 pc2 systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
May 18 19:50:33 pc2 systemd[1]: Stopped LoRa Server.
May 18 19:50:33 pc2 systemd[1]: Started LoRa Server.
May 18 19:50:33 pc2 loraserver[6640]: time="2017-05-18T19:50:33+05:30" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.16.1-2-gdde77eb
May 18 19:50:33 pc2 loraserver[6640]: time="2017-05-18T19:50:33+05:30" level=info msg="setup redis connection pool" url="redis://localhost:6379"
May 18 19:50:33 pc2 loraserver[6640]: time="2017-05-18T19:50:33+05:30" level=info msg="connecting to postgresql"
May 18 19:50:33 pc2 loraserver[6640]: time="2017-05-18T19:50:33+05:30" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
May 18 19:50:33 pc2 loraserver[6640]: time="2017-05-18T19:50:33+05:30" level=info msg="connecting to application-server" ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
May 18 19:50:33 pc2 loraserver[6640]: time="2017-05-18T19:50:33+05:30" level=info msg="no network-controller configured"
May 18 19:50:33 pc2 loraserver[6640]: time="2017-05-18T19:50:33+05:30" level=info msg="backend/gateway: connected to mqtt server"
May 18 19:50:33 pc2 loraserver[6640]: time="2017-05-18T19:50:33+05:30" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"
[... cut ...]
May 18 19:51:31 pc2 loraserver[6659]: time="2017-05-18T19:51:31+05:30" level=info msg="backend/gateway: rx packet received"
May 18 19:51:31 pc2 loraserver[6659]: time="2017-05-18T19:51:31+05:30" level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic" data_base64="gCIiIiKgPQAwMDAwKSWBmA=="
May 18 19:51:34 pc2 loraserver[6659]: time="2017-05-18T19:51:34+05:30" level=info msg="backend/gateway: gateway stats packet received" mac=aa555a0000000101

brocaar, 2017-05-19:
Ai, that is no good. I'll investigate the issue and release a fix asap.

brocaar, 2017-05-19:
Would you be able to run the following commands?

```bash
# stop the loraserver process
systemctl stop loraserver

# load configuration
set -a
source /etc/default/loraserver

# start loraserver manually
loraserver
```

It seems that the stacktrace is not logged. When running `loraserver` manually, I expect it will return some more information.

Richaagarwalrn, 2017-05-19:
Today I got specific log related to panic situation
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: time="2017-05-19T12:25:51+05:30" level=info msg="backend/gateway: gateway stats packet received" mac=aa555a0000000101
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: panic: value method gateway.GPSPoint.Value called using nil *GPSPoint pointer
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: goroutine 42 [running]:
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: panic(0x907540, 0xc42032c6e0)
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: /usr/local/go/src/runtime/panic.go:500 +0x1a1
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: github.com/brocaar/loraserver/internal/gateway.(*GPSPoint).Value(0x0, 0x9085c0, 0x0, 0xc4203176a0, 0x1)
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: <autogenerated>:1 +0xd4
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: database/sql/driver.defaultConverter.ConvertValue(0x9085c0, 0x0, 0x993020, 0xc420328b60, 0x0, 0x0)
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: /usr/local/go/src/database/sql/driver/types.go:217 +0xd0
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: database/sql.driverArgs(0x0, 0xc420317b60, 0x8, 0x8, 0xc420317a01, 0x0, 0x1100000001, 0xc4202be718, 0x0)
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: /usr/local/go/src/database/sql/convert.go:36 +0x8cf
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: database/sql.(*DB).exec(0xc4201426e0, 0x9bdee0, 0xc3, 0xc420317b60, 0x8, 0x8, 0xc420328b01, 0x0, 0x0, 0x0, ...)
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: /usr/local/go/src/database/sql/sql.go:1031 +0x3ea
May 19 12:25:51 richa-Lenovo-G40-80 loraserver[7691]: database/sql.(*DB).Exec(0xc4201426e0, 0x9bdee0, 0xc3, 0xc420317b60, 0x8, 0x8, 0x0, 0x0, 0x0, 0x0)
May 19 12:25:51 richa-Lenovo-G40-80 systemd[1]: loraserver.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
May 19 12:25:51 richa-Lenovo-G40-80 systemd[1]: loraserver.service: Unit entered failed state.
May 19 12:25:51 richa-Lenovo-G40-80 systemd[1]: loraserver.service: Failed with result 'exit-code'.
May 19 12:25:51 richa-Lenovo-G40-80 systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
May 19 12:25:51 richa-Lenovo-G40-80 systemd[1]: Stopped LoRa Server.
May 19 12:25:51 richa-Lenovo-G40-80 systemd[1]: Started LoRa Server.

Richaagarwalrn, 2017-05-19:
after executing these command
# stop the loraserver process
systemctl stop loraserver

# load configuration
set -a
source /etc/default/loraserver

# start loraserver manually
loraserver

**Log**
pc2@pc2:~$ journalctl -u loraserver -f -n 100
-- Logs begin at Fri 2017-05-19 19:22:19 IST. --
May 19 20:43:15 pc2 loraserver[7803]: panic: value method gateway.GPSPoint.Value called using nil *GPSPoint pointer
May 19 20:43:15 pc2 systemd[1]: loraserver.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
May 19 20:43:15 pc2 systemd[1]: loraserver.service: Unit entered failed state.
May 19 20:43:15 pc2 systemd[1]: loraserver.service: Failed with result 'exit-code'.
May 19 20:43:15 pc2 systemd[1]: loraserver.service: Service hold-off time over, scheduling restart.
May 19 20:43:15 pc2 systemd[1]: Stopped LoRa Server.
May 19 20:43:15 pc2 systemd[1]: Started LoRa Server.
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.16.1-2-gdde77eb
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="setup redis connection pool" url="redis://localhost:6379"
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="connecting to postgresql"
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="connecting to application-server" ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="no network-controller configured"
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="backend/gateway: connected to mqtt server"
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="backend/gateway: subscribing to stats topic" topic="gateway/+/stats"
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="migrated node-sessions to new format" errors=0 migrated=0
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="applying database migrations"
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="migrations applied" count=0
May 19 20:43:15 pc2 loraserver[7827]: time="2017-05-19T20:43:15+05:30" level=info msg="starting api server" bind="0.0.0.0:8000" ca-cert= tls-cert= tls-key=
May 19 20:43:16 pc2 loraserver[7827]: time="2017-05-19T20:43:16+05:30" level=info msg="node-session deleted" dev_eui=2222222222222222
May 19 20:43:16 pc2 loraserver[7827]: time="2017-05-19T20:43:16+05:30" level=info msg="node-session saved" dev_addr=22222222 dev_eui=2222222222222222
May 19 20:43:20 pc2 loraserver[7827]: time="2017-05-19T20:43:20+05:30" level=info msg="backend/gateway: rx packet received"
May 19 20:43:20 pc2 loraserver[7827]: time="2017-05-19T20:43:20+05:30" level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic" data_base64="gBERERGABAAwMDAwEG9O7w=="
May 19 20:43:25 pc2 loraserver[7827]: time="2017-05-19T20:43:25+05:30" level=info msg="backend/gateway: rx packet received"
May 19 20:43:25 pc2 loraserver[7827]: time="2017-05-19T20:43:25+05:30" level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic" data_base64="gBERERGABAAwMDAwEG9O7w=="
May 19 20:43:29 pc2 loraserver[7827]: time="2017-05-19T20:43:29+05:30" level=info msg="backend/gateway: rx packet received"
May 19 20:43:29 pc2 loraserver[7827]: time="2017-05-19T20:43:29+05:30" level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic" data_base64="gBERERGABAAwMDAwEG9O7w=="
May 19 20:43:35 pc2 loraserver[7827]: time="2017-05-19T20:43:35+05:30" level=info msg="backend/gateway: rx packet received"
May 19 20:43:35 pc2 loraserver[7827]: time="2017-05-19T20:43:35+05:30" level=error msg="processing rx packet error: get node-session error: node-session does not exist or invalid fcnt or mic" data_base64="gBERERGABAAwMDAwEG9O7w=="
May 19 20:43:42 pc2 loraserver[7827]: time="2017-05-19T20:43:42+05:30" level=info msg="backend/gateway: rx packet received"
May 19 20:43:42 pc2 loraserver[7827]: time="2017-05-19T20:43:42+05:30" level=info msg="packet(s) collected" dev_eui=1111111111111111 gw_count=1 gw_macs=aa555a0000000101 mtype=ConfirmedDataUp
May 19 20:43:42 pc2 loraserver[7827]: time="2017-05-19T20:43:42+05:30" level=info msg="rx info sent to network-controller" dev_eui=1111111111111111
May 19 20:43:42 pc2 loraserver[7827]: time="2017-05-19T20:43:42+05:30" level=warning msg="get gateways for macs error: expected 1 gateways, got 0" macs=[aa555a0000000101]
May 19 20:43:42 pc2 loraserver[7827]: time="2017-05-19T20:43:42+05:30" level=info msg="node-session saved" dev_addr=11111111 dev_eui=1111111111111111
May 19 20:43:42 pc2 loraserver[7827]: time="2017-05-19T20:43:42+05:30" level=info msg="backend/gateway: publishing tx packet" topic="gateway/aa555a0000000101/tx"
[... cut ...]
May 19 20:51:03 pc2 loraserver[8110]: time="2017-05-19T20:51:03+05:30" level=warning msg="get gateways for macs error: expected 1 gateways, got 0" macs=[aa555a0000000101]
May 19 20:51:03 pc2 loraserver[8110]: time="2017-05-19T20:51:03+05:30" level=info msg="node-session saved" dev_addr=11111111 dev_eui=1111111111111111
May 19 20:51:03 pc2 loraserver[8110]: time="2017-05-19T20:51:03+05:30" level=info msg="backend/gateway: publishing tx packet" topic="gateway/aa555a0000000101/tx"
May 19 20:51:03 pc2 loraserver[8110]: time="2017-05-19T20:51:03+05:30" level=info msg="node-session saved" dev_addr=11111111 dev_eui=1111111111111111
^C

brocaar, 2017-05-19:
When executing `loraserver`, you see the logs directly in your terminal (and the stacktrace would not be trunctated), so no need for a `journalctl -u loraserver -f -n 100`. I'll investigate this issue :-)

brocaar, 2017-05-22:
I think I've found the issue. You're compiling LoRa Server yourself using Go < 1.8. There seems to have been an issue regarding the handling of `nil` pointers to a struct and saving them to a database. This is where it goes wrong:

```go
// GPSPoint contains a GPS point.
type GPSPoint struct {
	Latitude  float64
	Longitude float64
}

// Value implements the driver.Valuer interface.
func (l GPSPoint) Value() (driver.Value, error) {
	return fmt.Sprintf("(%s,%s)", strconv.FormatFloat(l.Latitude, 'f', -1, 64), strconv.FormatFloat(l.Longitude, 'f', -1, 64)), nil
}
```

As in the `Gateway` struct, the `GPSPoint` can be `nil` the db library should just insert `NULL` into the db in that case. However it seems that the behaviour before Go 1.8 was that it was still calling the `Value` method which would blow up, resulting in the error `value method gateway.GPSPoint.Value called using nil *GPSPoint pointer`.

Could you please try compiling LoRa Server using Go 1.8 to see if that solves your issue?

For more details, see https://github.com/golang/go/issues/8415.

Richaagarwalrn, 2017-05-22:
Sorry for late reply.Yes I am using Go 1.7...will upgarde the go compiler and share the results.

Richaagarwalrn, 2017-05-24:
Yes it's working now. Thanks a lot.

closed 2017-06-19 by brocaar