brocaar/chirpstack-network-server #177: Where to set the application key and networksession key?

agau9527, 2017-05-23:
Hi, I'm new to Lora server.

My question is where to set the application key and networksession key? 

I have set up the packet_forwarder, lora_gateway bridge, lora app server, lora server.

And the lora server got the error message: Node-session does not exist or invalid fcnt or mic

I'm wondering whether if I do not set the appkey/nwskey correctly?

By the way, I can see the traffic on the app server web page. Can I see the detail message on that page?

brocaar, 2017-05-23:
Did you already login into LoRa App Server to setup your application and node? By default it is available on https://yourhost:8080/. The default login is `admin` / `admin`.

agau9527, 2017-05-24:
Hi, @brocaar 

Thanks for your help!

I set up the application and node on the web page with your directions.

The loraserver now can receive the message such as:

level=info msg="packet(s) collected" dev_eui=0102030405060708 gw_count=1 gw_macs=b827ebfffe3ea6d8 mtype=UnconfirmedDataUp

level=info msg="rx info sent to network-controller" dev_eui=0102030405060708

level=info msg="node-session saved" dev_addr=0d01023d **dev_eui=0102030405060708**

Another question is that can I see the detail node message(header, framepayload..) on the web page? I didn't find that.

brocaar, 2017-05-24:
No (not yet). For now the data is only exposed through https://docs.loraserver.io/lora-app-server/mqtt-topics/. Could you close this issue when your question has been answered?

closed 2017-05-24 by agau9527