brocaar/chirpstack-network-server #178: Make it possible to set a node to a fixed data-rate (no ADR)

brocaar, 2017-05-23:
This so that "mobile" (that should not use ADR) can still be configured to use a fixed data-rate (e.g. to avoid the usage of SF12).

danalvarez, 2020-05-26:
Hi, @brocaar, any update on this? Would be extremely useful for our application, mainly to be able to manually configure TX Power.

sagar-patel-sls, 2021-06-16:
Hi @brocaar any update on this feature?

brocaar, 2021-06-22:
Actually, it might be better to close this issue. Personally I don't think it is a good idea to expose this to end-users. E.g. when you are a network operator, you might not want to give your end-users control over these parameters.

Note that the pluggable ADR feature does allow you to implement this yourself and present alternative ADR options that can be selected by end-users, e.g. to set a device to a fixed data-rate. Note that the DR, TX power and NbTrans can be configured through your own ADR plugin implementation.

closed 2025-05-02 by brocaar