brocaar/chirpstack-network-server #179: No ACK from the Server since 0.17.1 (End device has a RN2483)

minggi, 2017-05-24:
Since the update to 0.17.1 I have a problem with the ACK from the Server. If I turn on the relax mode, the enddevice receive no ACK from the Server. If I turn off the relax mode, everthing work fine, as long as the counter is correct. The enddevice has a RN2483 as LoRa modem. The distance between the gateway and the end device is around 4km. We work with the ABP mode.
The whole system was working with the older Loraserver version.
Has anybody an enddevice with RN 2483 and 0.17.1 installed?
I have no idea how I can debug the issue.

Thanks for your help.

minggi, 2017-05-24:
Additional Info: I have no problem if I send a short payload (~4byte) But if the payload is around 30byte, I receive no ACK from the server. It works only for 3-7 payload.

brocaar, 2017-05-25:
What happens is that your node is probably sending a retransmit sometimes. The first uplink that was received by LoRa Server was processed and the counter was incremented at the side of LoRa Server. However, the ACK was not received by the node. Because of this, the node performed a re-transmit.

As LoRa Server notices that the counters are out-of-sync it will reset both the uplink and downlink counters on the server (as the counter mismatch could be because of a node-reset). However, the counter mismatch was not due to a node reset but because of packet-loss. Note that this only happens with the relax frame-counter mode turned on.

I’ll change the behaviour so that the uplink counter gets a reset (sync with what was received), but that it will not reset the downlink counter. Again please note that the relax frame-counter does not offer any security and that it is very likely that you are going to receive duplicated frames, as LoRa Server will not filter out the re-transmits!

commit 74433626 (2017-05-25): Fix relax frame-counter mode fCntDown reset on retransmit.

Closes #179.
closed 2017-05-25 by brocaar