brocaar/chirpstack-network-server #180: What happens when we lose Redis

siscia, 2017-05-24:
It happened to us to lost our Redis database.

It is quite an issues because to get back the connectivity we need to re-do a join.
It is this the expect behaviour?

For us is quite expensive to do a join, there is a way to store session related information in postgres and maybe use redis as cache?

(Read from redis by default,  if you don't find the session key go look in postgres?)

I may implement myself this feature if necessary.

brocaar, 2017-05-24:
Please note that Redis has various ways to persist data periodically to disk, I think that might be something to look at: https://redis.io/topics/persistence. Storing everything in PostgreSQL would raise the question "what happens when we lose PostgreSQL" 😉

siscia, 2017-05-24:
Hi :)

Redis is a marvelous piece of engineering.

I am more concerned from an operational point of view.

I do agree that, in theory, the problem is specular however:

1) it is quite more likely that, for any application, Postgres would hold some amount of business valuable data.

2) the focus of postgres is to do not lose data while being available

3) there are several business offering postgres as a managed service

Given this factors I tend to say that if something is gonna be missing between postgres and redis, most likely it will be redis.

Another, stronger point in my humble opinion, is that now I have to worry about two database, both postgres and redis, I definitely prefer to worry only about one.
If something goes south I may be ok to even receive the same message twice, I may cope with increased load on pg, I might even be ok to do not receive messages anymore. 
What I definitely cannot afford is to re-join my sensors.

In the redis community it is very usual not to encourage to use redis itself as a database and personally I tend to agree.

The biggest cost in software is the operational part and we should try to minimize it. Since the project is still young I believe we should consider this point as soon as possible.

This is a quite important architecture decision and I believe that whoever is somehow involved in the project should be ok with that.

What I can bring on the table is my experience in operating in an environment where a sensors are definitely not easy to reach, m

----- Messaggio originale -----

siscia, 2017-05-24:
What I can bring on the table is my experience in operating in an environment where a sensors are definitely not easy to reach, my concerns about the actual architecture and my developing capacity if necessary.

I hope to have a interesting conversation on this issue.

Thanks so much for your work,

Simone

brocaar, 2017-05-24:
I think you're addressing an interesting point. However putting Redis as a cache between PostgreSQL rather than a node-session storage would have a big impact on the performance, as it requires LoRa Server to write to PostgreSQL every time it receives a frame to make sure the session is in sync. Please note that Redis stores not only the keys of a node, it stores also the state of the nodes (activated channels, adr parameters, pending mac-commands, mac-command queues, pointers from DevAddr -> DevEUI). 

I don't think that PostgreSQL would be ideal for this kind of usage / data store and I think that implementing PostgreSQL as a fall-back (+ all the syncing between Redis and PostgreSQL) will add quite some complexity to the code.

> 3) there are several business offering postgres as a managed service

https://redislabs.com/products/redis-cloud/deployments/

Could you share more details about what happened to your Redis database / cluster?

reixd, 2017-05-24:
I agree with @siscia to use psql as the persistent storage unit and redis as a cache or for runtime stuff. Postgress could be used to store a shadow copy of the node keys and join data. 
As @siscia wrote, the loraserver would query redis looking for the node data, on empty response it would query psql and if no data found then behave accordingly.
The delays are negligible and in the IoT world waiting for a couple of ms more should not be an issue ;)

I propose to use a wrapper object (like a proxy) around the redis and psql management object. This wrapper will then be used to perform all queries. Internally the logic would then decide which DB to query and react to errors accordingly. Explicit functions to perform direct operations on a specific DB could be also be provided. 
A global config flag could be used to enable/disable psql as the persistent storage.

Overhead is also minimal and complexity is encapsulated into a DB management object but would guarantee a consistent code. Testing could be also simpler.
This change would also allow use different DB backends in the future :D
 
One problem I see is the management of the psql DB structure. It is not so flexible as with redis though.

reixd, 2017-05-24:
> node-session storage would have a big impact on the performance, as it requires LoRa Server to write to PostgreSQL every time it receives a frame to make sure the session is in sync. 

There is no need to write every operation to psql. There are different techniques to achieve this:

- Flush node data to psql every $minutes
- Use a management object to write to psql with a different thread in the backend
- connect to both dbs directly and write the data
- wait for an sql commit synchronously
- etc.

There are solutions ;) 

> Please note that Redis stores not only the keys of a node, it stores also the state of the nodes (activated channels, adr parameters, pending mac-commands, mac-command queues, pointers from DevAddr -> DevEUI).

this must not be really changed. For example on cache miss the psql will be queried, on cache hit the data is already stored in the redis db. Updating the persistent storage and the cache to have the same data is a bit tricky but not impossible. A way could be to have a LRU cache.

ivajloip, 2017-05-24:
Hi,

I am not sure how easy that change will be. My feeling is that it will not be all that easy (changing the code and fixing a bunch of tests).

At the same time I agree that it will have a performance impact on the whole system - not because of the reads, but because of the writes. Some of the proposed solutions could mitigate this to a great extend, but they will add a lot of complexity.

Could you explain why the redis persistence solutions won't work for you?

siscia, 2017-05-24:
Hi all, 
thanks for participating in this discussion and bring all this interesting points.

## About the performance - safety trade-off.

A good quote from the erlang community is that "there is not such thing as fast, there is only fast enough".

Myself, I have ping in order of half second to my gateway (GPRS), in this context some millisecond more waiting for postgres is not going to make any difference at all.

On top of this once we are getting serious about the whole architecture Redis is going to run on a dedicate host, different from the one where is running the lora-stuff.
In this context you are immediately going to get the network latency.
Once you factor in the network latency I don't think that update a counter in postgres vs updating the same counter in redis is going to be order of magnitude slower.
Especially if the table is fairly small (<100K nodes seems reasonable to me) and if you have activated AOF for redis (hit the disk after every write.)
(Then there is also the MQTT latency)

Factoring in everything in a reasonable safe architecture I don't see the update of the frameCounter as an huge issues.
(If you gentlemen don't agree with me we can run some test, this is just theory...)

## About the code complexity

The most expensive part of software is the operational part (running it) and not the developing part.
Given this assumption (which is true for long running software) I believe than any of us will prefer code reasonable more complex and less services necessary.
More things are in and more often they will break, in an unexpected way.
If things can be restart automatically without loosing data this is not an huge issues, but still.

## Degrees of Redis

From my point of view I believe we can have these degrees of "Redis-necessities"

1. Redis absolutely necessary, even to keep track of the nodes (where we are now)
2. Redis necessary only to break tie about what gateway received the best message (without redis we canno't safely receive messages, but we do not need a re-join)
3. Redis useful only to lower the load on the persistente DB
4. Redis unnecessary

Clearly 4 is the more desiderable, but it will be enough to get it down to 2 IMHO.

brocaar, 2017-05-24:
Before continuing this discussion, I would really like to know why the existing Redis persistence solutions won't work for you.

siscia, 2017-05-24:
There are several reason why I am pushing this issues.

1. We now have two critical point of failure, if either PG o Redis fails everything stop working, I would rather have a single POF

2. Redis is not made as primary data store, its persistence features are nice addons but not fundamental. The AOF write, by default, flush every second, it is not bad but it is not good neither. Making it flush at every command will make Redis slower than PG

I just feel uneasy, especially for storing data that are necessary for me to actually communicate with the remote nodes.

brocaar, 2017-05-25:
I appreciate your input, but at this stage I'm not yet convinced it is worth the effort to make these changes.

When implementing it so that PostgreSQL becomes a fallback, I don't think keeping Redis as a cache makes any sense, because every write to Redis would require a write to PostgreSQL. However, given the kind of data and the kind of operations performed on this data, I don't think PostgreSQL is a good fit for this data / I prefer to keep this data in Redis as it makes it easier for me to maintain the code.

This Redis to PostgreSQL refactor would cost me a lot of time (that I prefer to invest in other things), makes the code more complex and harder to maintain and LoRa Server less performant. There are many ways to setup Redis so that all data is persisted or to setup Redis as a cluster so that all data is stored in a redundant way (master / slave) over multiple servers.

Of course your Redis cluster could fail, but so could your PostgreSQL database, MQTT broker, LoRa Gateway Bridge, LoRa Server, LoRa App Server or your own application handling the data. However with good monitoring (and alerting) and a backup strategy, these risks can be limited to the minimum.

siscia, 2017-05-25:
I agree that the decision between the trade-offs is not simple.

Thanks to have addressed my points :) 

Simone

closed 2017-06-17 by siscia

stffabi, 2020-03-31:
I came across this issue after having found out that our LoRa server got useless after having deleted the Redis server. I also think this is a major drawback especially it's a major problem if the documentation states [1](https://www.chirpstack.io/network-server/install/requirements/#redis-database), [2](https://www.chirpstack.io/application-server/install/requirements/#redis-database)

> ChirpStack Network Server stores all non-persistent data into a Redis datastore.

> ChirpStack Application Server stores all non-persistent data into a Redis datastore

So Redis contains clearly not only non-persistent data, as already pointed out in this discussion it contains important persistent data.
Therefore we should at at least adjust the documentation and make this absolutely clear.

brocaar, 2020-03-31:
@stffabi I have updated the docs in the above commit.

stffabi, 2020-03-31:
@brocaar thanks for updating the docs.