brocaar/chirpstack-network-server #183: MQTT down, then need to restart lora{-app-}server ?

siscia, 2017-06-13:
Hi All,

we suffer an outage and I would like to explore what happened together.

On Saturday our MQTT server broke down (investigating also here what happened).

I restarted the MQTT and immediately our dashboard and accessory services connected to the server and we hoped for the best.

However, our gateways still aren't able to actually connect to the server. Nor we could see our messages.
Let me explain.

The gateways do not receive messages from the server.
After 5 times they do not receive any messages they simply shut down and restart, and again they wake up do not receive any message from the server and so on and so forth on loop.

I solved the whole problem by:
1. **BACKUP REDIS**
2. Restart loraserver
3. Restart lora-app-server

Why did I need to manually restart the two services ?

siscia, 2017-06-13:
Actualy I also needed to restart the gateway-bridge.

It is expected?

brocaar, 2017-06-19:
This is not expected behaviour and I think this should be handled by the MQTT library. Let me check this. Please note that the reconnect is default behaviour of the MQTT library: https://github.com/eclipse/paho.mqtt.golang/blob/master/options.go#L97.

siscia, 2017-06-19:
Hi,

Dont Know what to say.

Honestly I didn't re-test it.

Consider also that the MQTT server was down for quite a while (days).

Cheers :)

----- Messaggio originale -----

brocaar, 2017-06-19:
Please note that when the MQTT broker is down for quite a while, it can take up to 10 minutes (by default) before the MQTT client reconnects (I suppose they increase the retry time each time). See also https://github.com/eclipse/paho.mqtt.golang/blob/master/options.go#L96.

brocaar, 2017-06-19:
This issue could be related to this issue (which has been resolved ~ 17 days ago, so this has not yet been integrated in LoRa Server / App Server / Gateway Bridge): https://github.com/eclipse/paho.mqtt.golang/issues/96.

brocaar, 2017-06-19:
I've just updated all master branches with the latest mqtt library. I'll do a test tonight to see what happens when I shutdown Mosquitto and start it again tomorrow morning.

siscia, 2017-06-19:
Great, thank you :)

brocaar, 2017-06-22:
I've been testing with the updated MQTT library and after a long downtime (>8 hours) it successfully reconnects again. I'll release an update soon with the library update included :-)

siscia, 2017-06-22:
Fantastic!

Also, before the upgrade it didn't reconnect, right?

Are we sure that the problem was there?

brocaar, 2017-06-23:
Yes, I could confirm the issue before. I've just released a new version of LoRa Server. Please let me know if works better. If so, I'll release an update of the LoRa Gateway Bridge and LoRa App Server next week.

brocaar, 2017-06-23:
Hmm... there seems to be an other issue. After the new release I'm sometimes still experiencing a reconnection issue. It looks like something weird is going on during the re-connection. I'm currently looking at what is going over the wire, and it seems the MQTT client is switching the protocol version sometimes during a re-connect. Because of this the server is rejecting the connection. 

MQTT library (I've added this logging myself)
```
# Succesful (re)connect
conn bytes: 00044d5154540402001e0000
packet: CONNECT: dup: false qos: 0 retain: false rLength: 12 protocolversion: 4 protocolname: MQTT cleansession: true willflag: false WillQos: 0 WillRetain: false Usernameflag: false Passwordflag: false keepalive: 30 clientId:  willtopic:  willmessage:  Username:  Password:

# rejected reconnect
conn bytes: 00064d51497364700302001e0000
packet: CONNECT: dup: false qos: 0 retain: false rLength: 14 protocolversion: 3 protocolname: MQIsdp cleansession: true willflag: false WillQos: 0 WillRetain: false Usernameflag: false Passwordflag: false keepalive: 30 clientId:  willtopic:  willmessage:  Username:  Password:
```

Mosquitto
```
# Succesful (re)connect
1498223362: New connection from 127.0.0.1 on port 1883.
1498223362: New client connected from 127.0.0.1 as 521c9c34-327a-4da6-9478-93844475d7df (c1, k30).
1498223362: Sending CONNACK to 521c9c34-327a-4da6-9478-93844475d7df (0, 0)
1498223362: Received SUBSCRIBE from 521c9c34-327a-4da6-9478-93844475d7df
1498223362: 	gateway/+/rx (QoS 2)
1498223362: 521c9c34-327a-4da6-9478-93844475d7df 2 gateway/+/rx
1498223362: Sending SUBACK to 521c9c34-327a-4da6-9478-93844475d7df
1498223362: Received SUBSCRIBE from 521c9c34-327a-4da6-9478-93844475d7df
1498223362: 	gateway/+/stats (QoS 2)
1498223362: 521c9c34-327a-4da6-9478-93844475d7df 2 gateway/+/stats
1498223362: Sending SUBACK to 521c9c34-327a-4da6-9478-93844475d7df

# Rejected reconnect
1498223451: New connection from 127.0.0.1 on port 1883.
1498223451: Sending CONNACK to 127.0.0.1 (0, 2)
1498223451: Socket error on client <unknown>, disconnecting.
```

brocaar, 2017-06-23:
I've opened an issue for this https://github.com/eclipse/paho.mqtt.golang/issues/116.

siscia, 2017-06-23:
Maybe it is a mosquitto issues?

brocaar, 2017-06-23:
No, I think the MQTT client should not downgrade the MQTT protocol version from 3.1.1 to 3.1 after a successful connection. It is already known then that that protocol version was working and that it is likely that connection itself was interrupted / the MQTT broker is temporarily down.

brocaar, 2017-06-28:
The issue that I found has been fixed in the MQTT client library. Once the first connection has been established, the MQTT protocol version won't be downgraded during a failed re-connect. Hopefully this really solves the issue.

I'll update all components with this update MQTT client library.

closed 2017-06-28 by brocaar

brocaar, 2017-06-28:
I've released an update for all components :-)

siscia, 2017-06-28:
Wonderful :) 

Thanks for your work...

Now we should be share that the problem doesn't arise if also postrges or redis suffer a network partition from the applications...

Also having some kind of test for this would be quite interesting... Something like chaos testing or stuff... Are there any services that offer chaos testing for open source components?

commit e04d5cd7 (2017-06-28): Update MQTT vendor. Update changelog + bump version.

Closes #183.