brocaar/chirpstack-network-server #184: Manage reconnection to the MQTT server

siscia, 2017-06-14:
Referring to #183 

I noticed that the disconnect option is not managed.

I propose to change https://github.com/brocaar/loraserver/blob/master/internal/backend/gateway/backend.go#L205
 `Backend.onConnectionLost` in such a way to try to reconnect to the MQTT server.

Also, I will suggest changing the several `token.Wait()` into `token.WaitTimeout()` and eliminate the sleep (necessary in the case of network partition).
Also, I believe it could be a good idea to use exponential backoff, it shouldn't be an issue but you are never too careful.

I guess that this changes should be propagated in every package.

Thoughts ?

siscia, 2017-06-14:
A further thought.

The reconnection logic seems to be already handled by paho, also in a well-done way: https://github.com/eclipse/paho.mqtt.golang/blob/master/client.go#L276

However, this logic triggers only if we set the AutoReconnect flag. That I don't see us doing.

A quicker fix could be to simply set the above flag and forget about everything.
Or we could implement our custom logic in the `ConnectionLostHandler`

Implementing our custom logic is more prone to errors, however it is quite simple to do, but gave us more control.

brocaar, 2017-06-19:
Please note that `AutoReconnect` is set to `true` by default. https://godoc.org/github.com/eclipse/paho.mqtt.golang#NewClientOptions. When the MQTT broker is down for a long time, the default reconnection logic might take up to 10 minutes (because of the backoff). This might explain why reconnecting did not work immediately.

siscia, 2017-06-19:
I am talking of more than 8 hours of time from when I restarted the MQTT and when I finally restarted the actual Lora services.

----- Messaggio originale -----

brocaar, 2017-06-19:
I'm closing this issue in favour of #183 to keep the discussion at one place.

closed 2017-06-19 by brocaar