brocaar/chirpstack-network-server #186: unknown MType: UnconfirmedDataDown

mikumaycry, 2017-06-20:
```
time="2017-06-20T11:02:49+08:00" level=error msg="processing rx packet error: unknown MType: UnconfirmedDataDown" data_base64="YMi+tQYgvwC+tdEN"
```
While I repeat sending confirmed HEX msg every 20 seconds to server, the loraserver logs somes logs ***unknown MType: UnconfirmedDataDown***. 
Maybe the downlink msg is published to ***gateway/+/rx*** topic，but  I can't find out where the msg comes from.

yzk0281, 2017-06-20:
Yeah, same problem occured here, it seems that it is the gateway send the down data back to server.

brocaar, 2017-06-20:
Are you using multiple gateways?

yzk0281, 2017-06-20:
yes

yzk0281, 2017-06-20:
Is Multiple gatways have anything to do with this problem?
loraserver only deal gateway/+/rx topic, UnconfirmedDataDown msgs only send on topic gateway/+/tx.

brocaar, 2017-06-20:
In that case, I think that one gateway is receiving the downlink transmitted by an other gateway. Please note that the error itself is nothing to worry about. It is only indicating that the frame is rejected by LoRa Server, as it should be processed by the node.

yzk0281, 2017-06-20:
Great! This solved the problem puzzled my for a long while~~ thank you man.

mikumaycry, 2017-06-20:
Due to my node's antenna is not big enough to afford high bandwidth, we finally choose to use the same frequency for uplink and downlink.
I found that this problem would sometimes occur when server sends msg to node. When gateway is sending msg to node, the downlink msg is received by the gateway antenna before TX process finished.
I wonder if I can directly filter those msg on gateway ? Because I install lora-gateway-bridge on gateway too, I might process those msg before they are sent to server.
Those msg's RSSI and SNR is different from normal msg. Here is a sample log:

```
JSON up: {"rxpk":[{"tmst":1455801100,"chan":3,"rfch":1,"freq":472.900000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":8.0,"rssi":-85,"size":41,"data":"gGw7MQaAAgAIt2Y+1DY17F9bZppwu6z/cvuL7gKRvU4XubwTSZpPmRw="}]}
INFO: [up] PUSH_ACK received in 1 ms
INFO: [down] PULL_RESP received  - token[0:0] :)

JSON down: {"txpk":{"imme":false,"tmst":1456801100,"freq":472.9,"rfch":0,"powe":14,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":12,"data":"YGw7MQYgAQBzQ8eq"}}
TX: 60 6C 3B 31 06 20 01 00 73 43 C7 AA

INFO: Received pkt from mote: 06313B6C (fcnt=1)
RX: 60 6C 3B 31 06 20 01 00 73 43 C7 AA

JSON up: {"rxpk":[{"tmst":1457956156,"chan":0,"rfch":1,"freq":472.300000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":-6.8,"rssi":-111,"size":12,"data":"YGw7MQYgAQBzQ8eq"}]}
```

mikumaycry, 2017-06-21:
I make a new test under long range, the RSSI and SNR was bad but still works. It's not suitable to filter  packets according to these two params. I would try to filter them by Message type on gateway.

closed 2017-06-21 by mikumaycry