brocaar/chirpstack-network-server #187: Multiple loraserver connected to a single lora-app-server

MadHyena, 2017-06-20:
Hello,

We have been trying to set up an architecture using multiple LoRa servers and connecting them to the same LoRa application server. We need that because we want to connect some gateways on the 915MHz band and others on the 868MHz band to the same application server.

Having more than one loraserver connected to the lora-app-server works in practice, but it seems like the lora-app-server can only be aware of one of the instances of loraserver; at least we couldn't find a way to make it work with the configuration files.
This leads to some minor inconveniences like not having frame logs in the server GUI.

Is there any way to do this in the current state of the project that we might have missed?
If so, we would be grateful for any help in that regard, and if not, is that feature on the development roadmap?

Thank you in advance.

brocaar, 2017-06-22:
This is currently not possible. I would love to improve this, however with the upcoming LoRaWAN 1.1 specs and roaming feature and join-server, I think there might be some changes that would facilitate your needs.

From what I understood, a join-server will be introduced which will provide the network-server (LoRa Server) with the information about which application-server (LoRa App Server) should receive the payloads. That way you could use multiple application-servers with one loraserver, or use multiple loraserver instances with a single lora-app-server.

However, the specs are not made public yet for this by the LoRa Alliance.

MadHyena, 2017-06-22:
Okay, I understand.

Thank you very much!

closed 2017-07-10 by brocaar

vikkymahdrey, 2018-02-08:
Hi Brocaar,

We have ISM installation bands for different countries but I want to use installation band in AS.
What would be the best band for AS?

Please help me on this.
Thanks