brocaar/chirpstack-network-server #188: How to chose the best gateway to send downlink?

CXOldStar, 2017-06-20:
### Hi, I have a question how server choose the gateway to send the downlink to class A device.

1. The situtation is that the server receives multiple uplinks which are send from the same deivce and forwarded by different gateways. 
2. The server will send the ack downlink to the device. 
3. So what rules do the server choose the best gateway to send the downlink?

##### I think that the server should choose the best gateway based on the value of rssi. What do you think about it?

mikumaycry, 2017-06-21:
You can find collect.go in loraserver/internal/uplink folder, packets.go in loraserver/internal/models folder, loraserver will collect and sort data, responses to the one with best RSSI and SNR.

CXOldStar, 2017-06-22:
@mikumaycry Thank you very much! Do I understand the rules right?
- If both the values of SRN are equal or more than 5, the loraserver sorts gateways by rssi.
- In other hand, the loraserver sorts gateways by the value of SRN.

Can you explain why threshold value is 5 not the other value?

How loraserver sort gateways to keep the downlink recevied successful in the below situtation?
- loraserve receives uplinks from two gateways.
- Gateway_1, one of the gateways,has a good value of SNR but a worse value RSSI, such SNR is 8 and RSSI is -100
- Gateway_1, one of the gateways,has a good value of RSSI but a worse value SNR, such SNR is 2 and RSSI is -40

brocaar, 2017-06-22:
Please see https://github.com/brocaar/loraserver/issues/78#issuecomment-244065450 for the reason behind this decision.

closed 2017-07-15 by brocaar