brocaar/chirpstack-network-server #192: LoRaWAN node versioning?

brocaar, 2017-06-26:
I would love some feedback regarding node versioning. The LoRa Alliance has just updated the LoRaWAN Regional Parameters. In this document the TXPower table was updated. From the changelog:

> expressed all powers either as EIRP or as conducted power depending on regions

This itself is not an issue. The big issue is that they also change the delta between TX Power indices. For example in the old situation going from TXPower `0` to `1` (EU band, but applies to some other ISM bands too) would mean a step of `-6dB`, this has been changed to steps of `-2dB`. Although this allows a finer control of the TX Power, this change is not backwards compatible.

This means that when LoRa Server implements these new delta's and when ADR is turned on, it thinks it is asking a node to lower its tx power by `-2dB` but this could (for older nodes) also mean `-6dB` and in case no link margin is left, could disconnect the node.

I've asked this question to somebody of the LoRa Alliance and the feedback was:

> The assumption is that the server MUST know the device’s **LoRaWAN version** and the **regional param revision**. This information must be provided out-of-band when the device is registered.

This would mean that for each node, you need to know the:

* exact LoRaWAN version (e.g. 1.0.2)
* exact LoRaWAN Regional Parameters version (e.g. 1.0 or 1.0.2)

To me this doesn't feel the way to go, as most nodes don't even expose this information. Also as an end-user I would expect a LoRaWAN node to be working with a LoRaWAN network, without worrying about LoRaWAN 1.0 or 1.0.2 and the same for the LoRaWAN Regional Parameters. To me this should be handled by the protocol.

Please share your feedback / opinion / experience on this!

siscia, 2017-06-26:
That is quite an issue. 

I am not too much into the argument, however, if we ask a node to go down of 1 TXPowerIndex, can we understand if it went down of 6dB or of just 2dB?

In my opinion, the less the server know about the node the better.

I do agree with you that the user, at least the not very sophisticated, should not worry about those parameters.

I also agree that this should be part of the protocol.

On the other side, the fastest the transition ends the better off we will all be...

glederer007, 2017-06-30:
I like the answer you have got: _"This information must be provided out-of-band when the device is registered."_ Why should it be? As far as I know, it is not written anywhere in the LoRaWAN Specification By the way, what did he/she mean by _"out-of-band"_?

If I understand well, the "official" answer you have got makes all earlier certifications obsolete. Strictly speaking none of the "older" End Nodes could pass the current Certification Tests as - if I know well - ADR test are part of the Certification. But I hope I am wrong with this.
  
In my opinion ADR is still not very well defined in the LoRaWAN protocol and I think it won't be ever well defined. We have set up two pilot systems where after a couple of weeks of testing we left ADR and opted to let the End Node select with some "local intelligence" the best Data Rate / Power Index settings. The logic was very simple: at the deployment of the sensors, the sensors tried a confirmed message to the server with the highest TX power at the lowest data rate. If there was a confirmation, the TX power was lowered and the data rate was raised (all this automatically). This was repeated until there was no confirmation and then the values where set back to the latest confirmed combination. After deployment, during normal operation, the sensors were sending messages every 15 minutes. Every eighth message was sent as confirmed. If the message was confirmed everything was left as it was. If there was no confirmation, the power level was raised one step and the data rate was lowered. Of course this is not as sophisticated as ADR is, but after all it is doing its job. We are thinking of making a small modification to let the End Node test the possibility of lowering TX power and increasing data rate every once in a day, just to save some more battery power. 

I am afraid that the LoRa Alliance is not really able to keep up the pace with initiatives as Orne's open source server, or The Things Network, or Petr's open source lora server (the one written in Erlang) or the End Node experiments that all the enthusiasts (professionals and amateurs) are making. I hope the situation - and the attitude of the Alliance - will change soon, otherwise these kind of problems - like the TX power steps you mentioned - will arise more and more. Furthermore these problems need to be sorted ASAP as NB-IoT is almost here...

brocaar, 2017-07-01:
@glederer007 

> **As far as I know, it is not written anywhere in the LoRaWAN Specification By the way, what did he/she mean by "out-of-band"?**

Unfortunately, this is documented by the LoRaWAN specs now. See 4.2.2

> For each major version, end-devices may implement different minor versions of the frame format. The minor version used by an end-device must be made known to the network server beforehand using out of band messages (e.g., as part of the device personalization information).

With out-of-band they mean it is not covered by the LoRaWAN protocol to communicate. E.g. you would set these values manually when adding your node to LoRa App Server. 

> **The logic was very simple: at the deployment of the sensors, the sensors tried a confirmed message to the server with the highest TX power at the lowest data rate. If there was a confirmation, the TX power was lowered and the data rate was raised (all this automatically).**

You could even improve this by using the `LinkCheckReq` mac-command. LoRa Server will then return the link budget and the number of gateways that received your frame. `drSteps = (linkBudget - configurableMargin) / 3)` will give you the number of steps that the data-rate can be increased :-) When you can't increase the data-rate anymore, you could start lowering the tx-power. Basically you have the same logic as in LoRa Server, but then at the side of the node!

The good news is that I've been able to implement the LoRaWAN Regional Parameters 1.0.2 supporting both old and new nodes (with some tiny workarounds). But I also worry that these kind of future changes might cause a lot of fragmentations. The hardest part is to deal with nodes that are already out in the field that can't be easily upgraded.

brocaar, 2017-07-01:
@siscia 

> I am not too much into the argument, however, if we ask a node to go down of 1 TXPowerIndex, can we understand if it went down of 6dB or of just 2dB?

Unfortunately we can't. The good news is that I think in Europe all nodes are by default operating on TXPower 1 (14dBm). From TXPower 1 to TXPower 2 is less of an issue as the step is smaller.

To avoid too much deviation, the ADR engine now lowers the TXPower in steps of 1. So it might take a bit longer to reach the ideal TXPower, but at least it is safer when it is unknown what TXPower table the node implements.

siscia, 2017-07-01:
@glederer007 , @brocaar , 

Besides our technicality, who is on the board?

Brocaar is raising interesting concerns and everybody will need to modify its architecture to accommodate these changes.

There is no one in the Alliance board that may listen to the OpenSource community?

I believe that fragmentation won't help anybody. Nor us nor the more commercial oriented part of the Alliance (Good luck selling something that can run only on a specific, closed source web architecture!)

ivajloip, 2017-07-12:
I also agree that this change will make deployments and operations harder, especially if such changes continue to appear. In my personal opinion that kind of changes (that require out-of-band communication) are not good for the technology and for the community.

P.S: I was on vacation, so I am currently trying to catch up with the changes :)

borisbstyle, 2017-09-12:
I actually am struggling with this matter too. Most of my nodes are actually using 1.0.1 specs.
I would love to see the option for that

brocaar, 2017-09-12:
I think this might actually become a hard requirement for supporting LoRaWAN 1.1.

borisbstyle, 2017-09-12:
Is there a way to disable Tx power command changes in the current release?

closed 2017-11-19 by brocaar