brocaar/chirpstack-network-server #193: MQTT TLS options

temporaryaccount, 2017-06-28:
It is possible to define the ssl:// scheme for the MQTT server but there are no configuration options for setting the various TLS parameters like CA, client certificate, the TLS version and ciphers, etc.

siscia, 2017-06-28:
We are all busy, if you care about those feature I guess the fastest way is to implement them yourself.

https://github.com/brocaar/lora-gateway-bridge/issues/37

brocaar, 2017-06-28:
That might sound a bit harsh, but I'm sure that is not how @siscia is meaning it.

But it is true that this might not be our first priority right now, simply since there is much other work to do and there are only so many hours in a day 😉 So indeed it would be great if you could help with this if this is a feature that is important to you.

Please note that in the next release, there will be an option to configure a CA certificate. See: https://github.com/brocaar/loraserver/commit/136f43480122dfbf04df35fc7b14f317627d9d3d

minggi, 2017-06-28:
First two weeks of vacation 😉 After that I may have time.
Question: Only one certificate for all clients or one seperate certificate for each client ?

temporaryaccount, 2017-06-28:
@brocaar

> That might sound a bit harsh, but I'm sure that is not how @siscia is meaning it.
No worries, I'm very grateful to you guys for doing this work and sharing it.

> So indeed it would be great if you could help with this if this is a feature that is important to you.
For now I bypassed the issue by just *not* using client certificates (and relying only on password) and also using a real certificate signed by a trusted CA instead of a local CA, so it's not *that* important for me at this stage.

@minggi

> Only one certificate for all clients or one seperate certificate for each client ?
I don't think I understand the question. Isn't there just one client which connects to one MQTT server? TLS connection can be made with or without a client certificate signed by the CA. The client certificate allows the server to authenticated the client. Without it only the client can authenticate the server.

closed 2017-11-19 by brocaar