brocaar/chirpstack-network-server #195: loraserver didn't collect all packets during DeduplicationDelay

mikumaycry, 2017-06-29:
While I was testing join request, I found that loraserver not always response to the best packet, the one which has best loraSNR and RSSI, and I try to log all packet receive time and those packets going through filtering.

I change the time format in main()
```
customFormatter := new(log.TextFormatter)
customFormatter.TimestampFormat = time.StampMicro
customFormatter.FullTimestamp = true
log.SetFormatter(customFormatter)
```
and print RXInfoSet before **collectAndCallOnce** returns
```
sort.Sort(rxPacketWithRXInfoSet.RXInfoSet)
fmt.Println("************************")
fmt.Println(rxPacketWithRXInfoSet.RXInfoSet)
fmt.Println("************************")
return callback(rxPacketWithRXInfoSet)
```
the result was weird.
```
time="Jun 29 16:17:12.865765" level=info msg="backend/gateway: rx packet received"
time="Jun 29 16:17:12.866883" level=info msg="backend/gateway: rx packet received"
time="Jun 29 16:17:12.877196" level=info msg="backend/gateway: rx packet received"
************************
[{b827ebfffe552fa6 2017-06-29 08:17:12.848443226 +0000 UTC 4086483044 472100000 7 0 1 4/5 -105 -3 41 {LORA 12 125 0}}]
************************
```
those three packets have different loraSNR and RSSI, but same PHYPayload
loraSNR and RSSI  are as follow:

 frequency    RSSI    loraSNR
 472100  -105  -3.000000
 471900  -52  10.500000 
 471700  -100  -9.000000 

If all packet are collected, the second one should be the right one to response.Maybe the lock opration in backend.go or collect.go blocks.

brocaar, 2017-06-29:
What potentially could happen is that frames are received **after** LoRa Server started processing the collected frames due to network latency. This can be configured through the `--deduplication-delay` config flag and is by default set to `200ms`. Did you override this setting?

The set you're printing (`fmt.Println(rxPacketWithRXInfoSet.RXInfoSet)`) contains only one item so it seems that only one item was actually "collected".

brocaar, 2017-06-29:
One other question, do all gateways have an unique MAC? It would be hard for me to debug this, as I've been testing this before with multiple gateways, without any issue. Would you mind debugging this a bit further? E.g. the `collectAndCallOnce` should be called for each received frame. 

It will write the frame and then tries to acquire a lock so that the collected frames are collected only once. So the first time it will reach this part and sleep (https://github.com/brocaar/loraserver/blob/master/internal/uplink/collect.go#L84), after that other calls will return here (https://github.com/brocaar/loraserver/blob/master/internal/uplink/collect.go#L77).

You could also add some debug logging here:
https://github.com/brocaar/loraserver/blob/master/internal/uplink/uplink.go#L68

And here:
https://github.com/brocaar/loraserver/blob/master/internal/uplink/uplink.go#L55

And here:
https://github.com/brocaar/loraserver/blob/master/internal/backend/gateway/backend.go#L190

mikumaycry, 2017-06-29:
Thank you.I will make a further test with two gateways.
I came across this problem with one gateway.
The gateway producer told me that when the signal is too strong, a gateway might receive multiple same packets on different channel from one node and I have to filter them on server.
Those packets only differ in RXInfo and they are sent by a same gateway.

brocaar, 2017-06-29:
Ah, yes then it is a different problem. As there might be multiple instances of LoRa Server (performance / availability), a MAC + PHYPayload combination is only processed once. I'm not sure how critical this issue is, as I think it only happens when you have your node really close to your gateway (e.g. both devices are on your desk). 

For testing you could also lower the TXPower on your node, before join and tx.

mikumaycry, 2017-06-29:
Indeed I've tried different distance(close to antenna and far away from it) and TXPower(from 14db to 20db), this problem is not promised to happen. But when it happens, JOIN Request or Confirmed Msg will fail again and again, too much server resource would be taken.

closed 2017-09-04 by brocaar