brocaar/chirpstack-network-server #196: Insert probe into the messages

siscia, 2017-06-29:
Hi,

I was trying to come out with a decent strategy to Chaos test the whole system (loraserver + lora-app-server + lora-gateway-bridge + postgres + redis + MQTT server).

After our experience with MQTT I believe that it is something really useful and I would like to systematically approach the problem.

However is not as simple as it sounds, and it doesn't even sound simple...

One possible approach would be to count the messages that enter the system, count the messages that exist the system and being sure that the number is the one we are expecting while ChaosMonkey (or something similar) is running and the system is under load.

I believe I could go pretty far with this methodology but I am afraid it will be a little blind.

I am wondering how expensive would it be to add a unique identifier (+ timestamp) to each message that enters the system, maybe in the lora-gateway-bridge and pass along this identifier in every stage.
I am talking only about the MQTT messages not in Redis or PG.

Is it something approachable?

It is somehow emulable with a clever combination of frame counter, timestamps and gateway identifiers?

Still not even sure it is something worth add, I have never really set up a real test scenario, just exposing some idea and gathering feedback

brocaar, 2017-06-29:
But don't you already have unique identifiers for testing? When your gateway has a GPS, the timestamp field will be set by LoRa Gateway Bridge. Also each gateway is uniquely identified by its MAC address. I guess you potentially could combine this with other variable data (rx and tx parameters) to uniquely "fingerprint" mqtt messages as most of this information is also exposed by LoRa App Server.

siscia, 2017-06-30:
That may work, however a little issues may appear in cases like this one:

```
gateway/ac1dbeef00000001/rx {"rxInfo":{"mac":"ac1dbeef00000001","time":"2017-06-30T15:33:43.873153Z","timestamp":1920494228,"frequency":868300000,"channel":1,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-105,"loRaSNR":5.5,"size":23,"dataRate":{"modulation":"LORA","spreadFactor":12,"bandwidth":125}},"phyPayload":"QLM2FcaAKBQB57jLTg3ku+maccWorFM="}
application/1/node/0018b20000000a8d/rx {"applicationID":"1","applicationName":"Aspem","nodeName":"A8D","devEUI":"0018b20000000a8d","rxInfo":[{"mac":"ac1dbeef00000001","time":"2017-06-30T15:33:43.873153Z","rssi":-105,"loRaSNR":5.5,"name":"GW001","latitude":45.83285,"longitude":8.81284,"altitude":496}],"txInfo":{"frequency":868300000,"dataRate":{"modulation":"LORA","bandwidth":125,"spreadFactor":12},"adr":true,"codeRate":"4/5"},"fCnt":5160,"fPort":1,"data":"AZABAJ4IAQAMDQ=="}
gateway/ac1dbeef00000003/rx {"rxInfo":{"mac":"ac1dbeef00000003","time":"2017-06-30T15:33:43.873167Z","timestamp":3103927652,"frequency":868300000,"channel":1,"rfChain":1,"crcStatus":1,"codeRate":"4/5","rssi":-118,"loRaSNR":-10.8,"size":23,"dataRate":{"modulation":"LORA","spreadFactor":12,"bandwidth":125}},"phyPayload":"QLM2FcaAKBQB57jLTg3ku+maccWorFM="}
```

Same message but different gateway message...

brocaar, 2017-07-01:
Besides that from the `application/...` topic you have the gateway MAC addresses that are related to the other two `gateway/...` MQTT messages, you could also read the `fCnt` from the `phyPayload` to make sure they are related to the `fCnt` in the `application/...` topic.

brocaar, 2017-09-04:
With LoRaWAN 1.1 the communication between various components will be standardised (partly to facilitate roaming between different networks). This also includes which uplink / downlink meta-data is going to be exposed. Therefore I think now is not a good time to add any custom fields / "probes". I'll close this issue for now, lets take a look at it again after the LoRaWAN 1.1 release if it is still relevant by then :-)

closed 2017-09-04 by brocaar