brocaar/chirpstack-network-server #197: JWT error with curl?

PhatHub, 2017-06-30:
Okay... some back story.
1: I tried to use mosquitto_sub command (`mosquitto_sub -t "application/[appName]/#" -v`) to view what's published, but I couldn't find the App number. (I tried using the application name, and nothing is showing up.)
2: So I'm trying to find a list of applications with the gRPC API via curl (I copied the command from the web interface) but JWT isn't liking my authentication? (I used the base64 jwt random from my /etc/default config as the value for the Grpc-Metadata-Authorization object.)
`{"error":"authentication failed: jwt parse error: token contains an invalid number of segments","code":16}`

So am I missing something?
Thanks!

siscia, 2017-06-30:
The AppNumber is something like 1 or 2, well it actually depends on how many application you have.

Anyway if you log into the web view, where there is a list of all the Application you should see an ID next to the name of your application.

That is the number you have to use

:)

PhatHub, 2017-06-30:
Thanks! That helped with the mosquitto_sub side of things! (It's alive!) I was blind to that ID number... 

I'm still looking over JWT documentation to see where I'm going wrong with that. :( I have a friend that wants to interface via RESTful... (If you can't tell, I'm not a very webby kind of guy...)

siscia, 2017-06-30:
You need a JWT

Go on this page: https://jwt.io/

Craft your token: 

```
{
  "iss": "lora-app-server",
  "aud": "lora-app-server",
  "nbf": 1289566958,
  "exp": 1689653358,
  "sub": "your subject",
  "username": "your username"
}
```

Then put your secret on the field.

At this point the web page will spit out an encoded long string, on the left.

That is your JWT token :)

PhatHub, 2017-07-03:
ohhhh... Thank you! :)
For others that come onto this question later:
I finally understand [this page](https://docs.loraserver.io/lora-app-server/integrate/auth/) a little more.
And a python implementation of the [token generator...](https://pyjwt.readthedocs.io/en/latest/)

closed 2017-07-03 by PhatHub

PhatHub, 2017-07-03:
Another self-note for the API demo web page (and to let others know):
In `/api/` scroll down to the "Internal" section, and expand the `/api#!/Internal/Login` section.
This should generate the JWT token with just the username and password json, no token needed. The generated POST command example is what I need in order to have the server send me my temporary JWT token.
I guess I posted this question on the wrong project because this API was talked about (over and over again) on the [lora-app-server issues](https://github.com/brocaar/lora-app-server/issues?utf8=%E2%9C%93&q=jwt) board! :-)

JairoRotava, 2018-02-18:
I would like to share an example with more details about this, because I took some time trying to understand this even with all the help here. When using the API page of lora-app-server you need to provide the JWT TOKEN, not the JWT SECRET itself. In my case the JWT SECRET was apisecret. 

To create the token you go on this page: https://jwt.io/, change the payload field to 
```
{
	"iss": "lora-app-server",
	"aud": "lora-app-server",
	"nbf": 1489566958,
	"exp": 1589653358,
	"sub": "user",
	"username": "admin"
}
```

Update the secret field to your secret (in may case apisecret),then you get a token like this:

`eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJsb3JhLWFwcC1zZXJ2ZXIiLCJhdWQiOiJsb3JhLWFwcC1zZXJ2ZXIiLCJuYmYiOjE0ODk1NjY5NTgsImV4cCI6MTU4OTY1MzM1OCwic3ViIjoidXNlciIsInVzZXJuYW1lIjoiYWRtaW4ifQ.d-Cq1QgFgvrggiGXfFh7y8vGkRFboy8Fjk5ZZNmGcyM`

This is the token you should use for the JWT TOKEN field on the API page.