brocaar/chirpstack-network-server #200: Misleading instructions in https://docs.loraserver.io

glederer007, 2017-07-05:
I have found some (apparently) contradictory instructions on how to set databases in psql.
 
https://docs.loraserver.io/install/quick-install/ instructs the following:

_-- set up the users and the passwords (note that it is important to use single quotes)
create role loraserver with login password 'dbpassword';
create role loraappserver with login password 'dbpassword';

-- create the database for the servers
create database loraserver with owner loraserver;
create database loraappserver with owner loraappserver;

-- exit psql
\q_

On the other hand https://docs.loraserver.io/lora-app-server/install/debian/ states the following:

_-- create the loraserver_as user
create role loraserver_as with login password 'dbpassword';

-- create the loraserver_as database
create database loraserver_as with owner loraserver_as;

-- exit the prompt
\q_

And https://docs.loraserver.io/loraserver/install/debian/ :

_-- create the loraserver_ns user with password 'dbpassword'
create role loraserver_ns with login password 'dbpassword';

-- create the loraserver_ns database
create database loraserver_ns with owner loraserver_as;

-- exit the prompt
\q_

So which is the correct role and name of the databases:  _loraserver / loraappserver_ or _loraserver_ns / loraserver_as_ ?

brocaar, 2017-07-06:
The database name doesn't really matter, as long as it matches the configuration given to LoRa Server / LoRa App Server.

When you think it would be better to make everything consistent, would you mind making these changes (changing the db name for LoRa App Server to `loraserver_as` and for LoRa Server to `loraserver_ns`)? That would be appreciated :-)

glederer007, 2017-07-06:
Hello Orne,

I have corrected the mistyping in the loraserver installation guide. However I don't know what to fork to correct the Quick Install part of the project (https://docs.loraserver.io/install/quick-install/)

brocaar, 2017-07-06:
I just merged your PR :-) The "project-wide" documentation is in https://github.com/brocaar/docs.loraserver.io.

brocaar, 2017-07-15:
All has been merged :-)

closed 2017-07-15 by brocaar