brocaar/chirpstack-network-server #203: can't connect to with app_server

ckt1010, 2017-07-07:
Hi,
I use lateset code and got log below:

> loraserver_1     | time="2017-07-07T09:40:17Z" level=info msg="backend/gateway: subscribing to stats 
> topic" topic="gateway/+/stats"
> loraserver_1     | time="2017-07-07T09:40:17Z" level=info msg="migrated node-sessions to new format" errors=0 migrated=0
> loraserver_1     | time="2017-07-07T09:40:17Z" level=info msg="applying database migrations"
> loraserver_1     | time="2017-07-07T09:40:17Z" level=info msg="migrations applied" count=0
> loraserver_1     | time="2017-07-07T09:40:17Z" level=info msg="starting api server" bind="0.0.0.0:8000" ca-cert= tls-cert= tls-key=
> loraserver_1     | time="2017-07-07T09:40:18Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:18Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:18Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:18Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:19Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:19Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:19Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:20Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:20Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:20Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:20Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:21Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:21Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:21Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> app_server_1     | Starting Lora App Server
> app_server_1     | ./build/lora-app-server
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="starting LoRa App Server" docs="https://docs.loraserver.io/" version=0.10.1
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="connecting to postgresql"
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="setup redis connection pool"
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="handler/mqtt: connecting to mqtt broker" server="tcp://mosquitto:1883"
> mosquitto_1      | 1499420421: New connection from 172.20.0.9 on port 1883.
> mosquitto_1      | 1499420421: New client connected from 172.20.0.9 as 2169f9a7-901e-4993-8869-b98d22ab6977 (c1, k30).
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="connecting to network-server api" ca-cert= server="loraserver:8000" tls-cert= tls-key=
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="handler/mqtt: connected to mqtt broker"
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx"
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="applying database migrations"
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="migrations applied" count=0
> app_server_1     | time="2017-07-07T09:40:21Z" level=info msg="gwmigrate: migrating gateway data from LoRa Server"
> loraserver_1     | time="2017-07-07T09:40:21Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> loraserver_1     | time="2017-07-07T09:40:22Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> app_server_1     | time="2017-07-07T09:40:22Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."
> app_server_1     | time="2017-07-07T09:40:22Z" level=error msg="gwmigrate: list gateways error (will retry): rpc error: code = Unavailable desc = transport is closing"
> loraserver_1     | time="2017-07-07T09:40:22Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken EOF."

Is it because of I miss set "ca-cert= tls-cert= tls-key="? I try to use TLS key also, but still got:

> "transport: authentication handshake failed: EOF"; Reconnecting to {loraserver:8000 <nil>}"

sorry for this stupid question....

ckt1010, 2017-07-12:
It should grpc proxy problem. Still finding how to workaround it ......

brocaar, 2017-07-15:
It looks like an issue between the LoRa Server and LoRa App Server connection to me.

ckt1010, 2017-07-26:
Hi brocaar,
you mean it is known issue? Actaully I test it on London's server and without any proxy can fix this problem.

brocaar, 2017-07-26:
I think it is more a networking issue between the two instances. Please create a topic at https://forum.loraserver.io/ in case you would discuss this any further. I'm trying to move support questions to the forum to cleanup the GitHub issues a bit :-)

closed 2017-07-26 by brocaar